GlasgowEmbedded/glasgow #303: jtag-openocd behaviour doesn't match jtag-svf

jeremyherbert, 2023-02-06:
Apologies in advance if this is an upstream bug in openocd or ecppack.

I have an SVF file which is just a blinky for an ECP5 (LFE5U-12F) FPGA built with ecppack. If I use the `jtag-svf` applet, it loads on the FPGA fine:

```
$ glasgow run jtag-svf -M --port A --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 blink.svf 
I: g.device.hardware: building bitstream ID baa3e836c8b20784cde3bbff3b5fe82e
/home/jeremy/oss-cad-suite/bin/nextpnr-ice40: line 22: /home/jeremy/.config/yosyshq/fonts.conf: No such file or directory
I: g.cli: running handler for applet 'jtag-svf'
I: g.applet.interface.jtag_svf: port A voltage set to 3.3 V
W: g.applet.interface.jtag_svf: SVF: test vector did not reset DUT explicitly, resetting
```

However, if I try to use the `jtag-openocd` applet with the following configuration file:

```
adapter driver remote_bitbang
remote_bitbang port 33333
remote_bitbang host localhost

reset_config none separate

jtag newtap ecp5 tap -irlen 8 -expected-id 0x21111043

init
scan_chain
svf -tap ecp5.tap -quiet -progress blink.svf
shutdown
```

the FPGA does actually load the bitstream and start running (I can see the blinking LEDs), but openocd throws an error:

```
$ glasgow run jtag-openocd -M --port A --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 tcp:localhost:33333
I: g.device.hardware: building bitstream ID b6ca43076c531bba087e54cd1f647cdb
/home/jeremy/oss-cad-suite/bin/nextpnr-ice40: line 22: /home/jeremy/.config/yosyshq/fonts.conf: No such file or directory
I: g.cli: running handler for applet 'jtag-openocd'
I: g.applet.interface.jtag_openocd: port A voltage set to 3.3 V
I: g.applet.interface.jtag_openocd: socket: listening at tcp:localhost:33333
I: g.applet.interface.jtag_openocd: socket: new connection from [127.0.0.1]:51492
I: g.applet.interface.jtag_openocd: socket: connection from [127.0.0.1]:51492 lost
```

```
$ openocd -f flash.ocd 
Open On-Chip Debugger 0.12.0+dev-00059-g0b6f53e94-dirty (2023-01-29-02:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:33333
Info : remote_bitbang driver initialized
Info : Note: The adapter "remote_bitbang" doesn't support configurable speed
Info : JTAG tap: ecp5.tap tap/device found: 0x21111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x2)
Warn : gdb services need one or more targets defined
svf processing file: "blink.svf"
95%    Error: tdo check error at line 28
Error:     READ = 0x6401e10
Error:     WANT = 0x0000000
Error:     MASK = 0x000b000
Time used: 0m16s183ms 

svf file programmed failed
Info : remote_bitbang interface quit
```

Presumably there isn't anything wrong with the SVF file because the `jtag-svf` applet works fine - unless there are some SVF lines that are skipped in the `jtag-svf` applet.

[blink.svf.txt](https://github.com/GlasgowEmbedded/glasgow/files/10661891/blink.svf.txt) (.txt extension added to allow it to be uploaded)

jeremyherbert, 2023-02-06:
Hmm, I’m thinking this is because there is no openocd remote_bitbang command for a delay. The glasgow server is accepting packets as fast as possible, and so the delay in data packets is not transferred to the device due to the fact that the output data rate of the FPGA is slower than the input data rate from the USB interface.

jeremyherbert, 2023-02-07:
Indeed, that appears to be the issue. Something like this appears to fix the problem: https://review.openocd.org/c/openocd/+/7472

Basically, it just adds two commands to delay either 1ms or 1us. If the patch is accepted, then it should be straightforward to add some delays to the JTAG applet I think.

whitequark, 2023-02-07:
I wonder if maybe applying backpressure to the bitbang source is actually the correct approach here? Can we do this?

jeremyherbert, 2023-02-07:
I thought about this (both with some sort of XON/XOFF and a ‘checkpoint’ handshake at points in SVF execution), but I can’t think of a good way to send the back pressure indication to openocd without increasing the complexity of the remote_bitbang protocol (which I don’t think will fly). As I understand it, the original intention for the protocol was for use with simulations, so they probably want to keep it as simple as possible. 

Though, I think there is no reason that Glasgow needs to utilise the functionality in the linked patch - perhaps a different way would be more suitable given the extreme flexibility of the hardware. I mostly put the patch together to confirm that was indeed the problem, and also because I’d like to use it with another non-Glasgow device. For this applet, I think that maybe long term it would be better to use/create another adapter in openocd that also allows one to set the clock rate and maybe some other options - this could also include back pressure indication.

whitequark, 2023-02-07:
LGTM on adding support for this specific interface to `jtag-openocd`.

whitequark, 2023-10-21:
The upstream continues being unresponsive for seemingly no reason. :(

whitequark, 2023-12-17:
This has now been merged upstream: https://review.openocd.org/c/openocd/+/7472

jeremyherbert, 2024-03-31:
See also https://review.openocd.org/c/openocd/+/8191

whitequark, 2024-06-27:
@jeremyherbert I think this should be fixed now--could you retest please?

jeremyherbert, 2024-07-02:
I tried running the following command with the SVF file in the original post:

```
glasgow run jtag-svf -V 3.3 --port A --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 blink.svf
```

But I receive the error:

```
I: g.device.hardware: device already has bitstream ID 62239ba31f7697d2809a2646248c966a
I: g.cli: running handler for applet 'jtag-svf'
I: g.applet.interface.jtag_svf: port(s) A voltage set to 3.3 V
Traceback (most recent call last):
  File "/home/jeremy/.local/bin/glasgow", line 8, in <module>
    sys.exit(run_main())
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 939, in run_main
    exit(asyncio.new_event_loop().run_until_complete(main()))
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 707, in main
    return applet_task.result()
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 656, in run_applet
    return await applet.interact(device, args, iface)
  File "/home/jeremy/Downloads/glasgow/software/glasgow/applet/interface/jtag_svf/__init__.py", line 294, in interact
    await coro
  File "/home/jeremy/Downloads/glasgow/software/glasgow/applet/interface/jtag_svf/__init__.py", line 120, in svf_state
    path = {
KeyError: (<JTAGState.UNKNOWN: 'Unknown'>, <JTAGState.IDLE: 'Run-Test/Idle'>)
```

I also tried with a freshly compiled openocd and the openocd config file above:

```
Open On-Chip Debugger 0.12.0+dev-00634-g23c33e1d3 (2024-07-02-17:11)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:33333
Info : remote_bitbang driver initialized
Info : Note: The adapter "remote_bitbang" doesn't support configurable speed
Info : JTAG tap: ecp5.tap tap/device found: 0x41111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x4)
Warn : JTAG tap: ecp5.tap       UNEXPECTED: 0x41111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x4)
Error: JTAG tap: ecp5.tap  expected 1 of 1: 0x21111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x2)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
svf processing file: "blink.svf"
  0%  HDR	0;
(Above Padding command skipped, as per -tap argument)
  0%  HIR	0;
(Above Padding command skipped, as per -tap argument)
  0%  TDR	0;
(Above Padding command skipped, as per -tap argument)
  0%  TIR	0;
(Above Padding command skipped, as per -tap argument)
  0%  ENDDR	DRPAUSE;
  0%  ENDIR	IRPAUSE;
  0%  STATE	IDLE;
  0%  SIR	8	TDI  (E0);
  0%  			MASK (FFFFFFFF);
  0%  SIR	8	TDI  (1C);
  0%  				FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF);
  0%  SIR	8	TDI  (C6);
  0%  SDR	8	TDI  (00);
  0%  RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
  0%  SIR	8	TDI  (0E);
  0%  SDR	8	TDI  (01);
  0%  RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
[... 139 lines cut ...]
  File "/home/jeremy/Downloads/glasgow/software/glasgow/access/direct/demultiplexer.py", line 240, in read
    await self._in_tasks.wait_one()
  File "/home/jeremy/Downloads/glasgow/software/glasgow/support/task_queue.py", line 81, in wait_one
    await asyncio.wait(self._live, return_when=asyncio.FIRST_COMPLETED)
  File "/usr/lib/python3.10/asyncio/tasks.py", line 368, in wait
    raise ValueError('Set of coroutines/Futures is empty.')
ValueError: Set of coroutines/Futures is empty.
E: asyncio: Task was destroyed but it is pending!
task: <Task pending name='Task-21' coro=<JTAGOpenOCDApplet.interact.<locals>.forward_out() running at /home/jeremy/Downloads/glasgow/software/glasgow/applet/interface/jtag_openocd/__init__.py:159> wait_for=<Future pending cb=[Task.task_wakeup()]>>
```

Probe does work though:
```
glasgow run jtag-probe -V 3.3 --port A --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 scan
I: g.device.hardware: device already has bitstream ID 62239ba31f7697d2809a2646248c966a
I: g.cli: running handler for applet 'jtag-probe'
I: g.applet.interface.jtag_probe: port(s) A voltage set to 3.3 V
I: g.applet.interface.jtag_probe: shifted 32-bit DR=<11000010000010001000100010000010>
I: g.applet.interface.jtag_probe: shifted 8-bit IR=<10000000>
I: g.applet.interface.jtag_probe: discovered 1 TAPs
I: g.applet.interface.jtag_probe: TAP #0: IR[8] IDCODE=0x41111043
I: g.applet.interface.jtag_probe: manufacturer=0x021 (Lattice Semi.) part=0x1111 version=0x4
```

I am using the version `0.1.dev2093+g00ad180` with amaranth `0.5.0`.

whitequark, 2024-07-02:
@wanda-phi That sounds related to your changes, can you take a look?

jeremyherbert, 2024-07-03:
A small update - the openocd error was caused by me using an ECP5 25F to test rather than an ECP5 12F. Sorry, was just in a bit of a rush. 

Here is what a 12F looks like (working):

```
$ glasgow run jtag-openocd -V 3.3 --port A --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 tcp:localhost:33333
I: g.device.hardware: generating bitstream ID f4351d43ebcc1e42ec5196bfddf74d47
I: g.cli: running handler for applet 'jtag-openocd'
I: g.applet.interface.jtag_openocd: port(s) A voltage set to 3.3 V
I: g.applet.interface.jtag_openocd: socket: listening at tcp:localhost:33333
I: g.applet.interface.jtag_openocd: socket: new connection from [127.0.0.1]:59722
I: g.applet.interface.jtag_openocd: socket: connection from [127.0.0.1]:59722 lost
```

```
$ openocd -f flash.ocd
Open On-Chip Debugger 0.12.0+dev-00634-g23c33e1d3 (2024-07-02-17:11)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:33333
Info : remote_bitbang driver initialized
Info : Note: The adapter "remote_bitbang" doesn't support configurable speed
Info : JTAG tap: ecp5.tap tap/device found: 0x21111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x2)
Warn : gdb services need one or more targets defined
svf processing file: "blink.svf"
  0%  HDR	0;
(Above Padding command skipped, as per -tap argument)
  0%  HIR	0;
(Above Padding command skipped, as per -tap argument)
  0%  TDR	0;
(Above Padding command skipped, as per -tap argument)
  0%  TIR	0;
(Above Padding command skipped, as per -tap argument)
  0%  ENDDR	DRPAUSE;
  0%  ENDIR	IRPAUSE;
  0%  STATE	IDLE;
  0%  SIR	8	TDI  (E0);
  0%  			MASK (FFFFFFFF);
  0%  SIR	8	TDI  (1C);
  0%  				FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF);
  0%  SIR	8	TDI  (C6);
  0%  SDR	8	TDI  (00);
  0%  RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
  0%  SIR	8	TDI  (0E);
  0%  SDR	8	TDI  (01);
  0%  RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
  0%  SIR	8	TDI  (3C);
  0%  			MASK (0000B000);
  0%  SIR	8	TDI  (46);
  0%  SDR	8	TDI  (01);
  0%  RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
  0%  SIR	8	TDI  (7A);
  0%  RUNTEST	IDLE	2 TCK	1.00E-02 SEC;
  0%  			0000DCFFFFFFFFCDBDFFFFFF006CAC4C82E24282C21CB4624C8CB4AAACA26232045C2E4E860A00FF);
  0%  			00FF173300000000000000000000FF173300000000000000000000FF173300000000000000000000);
  0%  			FF173300000000000000000000FF173300000000000000000000FF173300000000000000000000FF);
  5%  			173300000000000000000000FF173300000000000000000000FF173300000000000000000000FF17);
  5%  			000000000000000000FF173300000000000000000000FF173300000000000000000000FF17330000);
  5%  			0000000000000000FF173300000000000000000000FF173300000000000000000000FF1733000000);
  5%  			000000000000FF173300000000000000000000FF173300000000000000000000FF17330000000000);
  5%  			00000000FF173300000000000000000000FF173300000000000000000000FF173300000000000000);
 10%  			000000FF173300000000000000000000FF173300000000000000000000FF17330000000000000000);
 10%  			00FF173300000000000000000000FF173300000000000000000000FF173300000000000000000000);
 10%  			173300000000000000000000FF173300000000000000000000FF173300000000000000000000FF17);
 10%  			3300000000000000000000FF173300000000000000000000FF173300000000000000000000FF1733);
 10%  			00000000000000000000FF173300000000000000000000FF173300000000000000000000FF173300);
 15%  			0000000000000000FF173300000000000000000000FF173300000000000000000000FF1733000000);
[... 100 lines cut ...]

Unfortunately the `jtag-svf` applet still gives the same error:

```
$ glasgow run jtag-svf -V 3.3 --port A --pin-tck 0 --pin-tms 3 --pin-tdi 2 --pin-tdo 1 blink.svf
I: g.device.hardware: generating bitstream ID 62239ba31f7697d2809a2646248c966a
I: g.cli: running handler for applet 'jtag-svf'
I: g.applet.interface.jtag_svf: port(s) A voltage set to 3.3 V
Traceback (most recent call last):
  File "/home/jeremy/.local/bin/glasgow", line 8, in <module>
    sys.exit(run_main())
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 939, in run_main
    exit(asyncio.new_event_loop().run_until_complete(main()))
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 707, in main
    return applet_task.result()
  File "/home/jeremy/Downloads/glasgow/software/glasgow/cli.py", line 656, in run_applet
    return await applet.interact(device, args, iface)
  File "/home/jeremy/Downloads/glasgow/software/glasgow/applet/interface/jtag_svf/__init__.py", line 294, in interact
    await coro
  File "/home/jeremy/Downloads/glasgow/software/glasgow/applet/interface/jtag_svf/__init__.py", line 120, in svf_state
    path = {
KeyError: (<JTAGState.UNKNOWN: 'Unknown'>, <JTAGState.IDLE: 'Run-Test/Idle'>)
```

wanda-phi, 2024-07-03:
Indeed, that is my bug. Could you test if #615 fixes it?

jeremyherbert, 2024-07-04:
Works great now! Thanks for the fix. Please feel to close this as necessary @whitequark

closed 2024-07-04 by whitequark