GlasgowEmbedded/glasgow #304: Synthesis for applets fails with YoWASP with: ERROR: Assert `ofs.is_open()' failed in ../yosys-src/passes/techmap/abc9_ops.cc:950.

VioletEternity, 2023-02-16:
I've installed Glasgow and YoWASP in a new virtual environment:

<details>
<summary>Shell transcript</summary>

```
C:\Users\Maya\Desktop\glasgow>python -m venv venv

C:\Users\Maya\Desktop\glasgow>venv\Scripts\activate

(venv) C:\Users\Maya\Desktop\glasgow>pip install ./software[toolchain]
Processing c:\users\maya\desktop\glasgow\software
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting amaranth@ git+https://github.com/amaranth-lang/amaranth.git
  Cloning https://github.com/amaranth-lang/amaranth.git to c:\users\maya\appdata\local\temp\pip-install-v05ktafr\amaranth_9c9a5d56b90b46b1af4f715b17b29de6
  Running command git clone --filter=blob:none --quiet https://github.com/amaranth-lang/amaranth.git 'C:\Users\Maya\AppData\Local\Temp\pip-install-v05ktafr\amaranth_9c9a5d56b90b46b1af4f715b17b29de6'
  Resolved https://github.com/amaranth-lang/amaranth.git to commit 7e3e10e7333630999c5b51440f0e5f0daef0358c
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting crcmod
  Using cached crcmod-1.7.tar.gz (89 kB)
  Preparing metadata (setup.py) ... done
Collecting libusb1>=1.8.1
  Using cached libusb1-3.0.0-py3-none-win_amd64.whl (140 kB)
Collecting pyvcd
  Using cached pyvcd-0.3.0-py2.py3-none-any.whl (23 kB)
Collecting bitarray
  Downloading bitarray-2.7.2-cp310-cp310-win_amd64.whl (116 kB)
     ---------------------------------------- 116.2/116.2 kB 3.3 MB/s eta 0:00:00
Collecting fx2>=0.11
  Using cached fx2-0.11-py3-none-any.whl (17 kB)
Collecting aiohttp~=3.8
  Downloading aiohttp-3.8.4-cp310-cp310-win_amd64.whl (319 kB)
     ---------------------------------------- 319.8/319.8 kB 4.9 MB/s eta 0:00:00
Collecting yowasp-nextpnr-ice40>=0.1
  Using cached yowasp_nextpnr_ice40-0.5.0.0.post399-py3-none-any.whl (71.9 MB)
Collecting amaranth-yosys
  Using cached amaranth_yosys-0.25.0.0.post67-py3-none-any.whl (740 kB)
Collecting yowasp-yosys>=0.11
  Using cached yowasp_yosys-0.26.0.0.post489-py3-none-any.whl (7.0 MB)
Collecting charset-normalizer<4.0,>=2.0
  Downloading charset_normalizer-3.0.1-cp310-cp310-win_amd64.whl (96 kB)
     ---------------------------------------- 96.5/96.5 kB ? eta 0:00:00
Collecting yarl<2.0,>=1.0
  Using cached yarl-1.8.2-cp310-cp310-win_amd64.whl (56 kB)
Collecting aiosignal>=1.1.2
  Using cached aiosignal-1.3.1-py3-none-any.whl (7.6 kB)
Collecting attrs>=17.3.0
  Downloading attrs-22.2.0-py3-none-any.whl (60 kB)
     ---------------------------------------- 60.0/60.0 kB 3.1 MB/s eta 0:00:00
Collecting multidict<7.0,>=4.5
  Downloading multidict-6.0.4-cp310-cp310-win_amd64.whl (28 kB)
Collecting async-timeout<5.0,>=4.0.0a3
  Using cached async_timeout-4.0.2-py3-none-any.whl (5.8 kB)
Collecting frozenlist>=1.1.1
  Using cached frozenlist-1.3.3-cp310-cp310-win_amd64.whl (33 kB)
Collecting yowasp-runtime~=1.1
  Using cached yowasp_runtime-1.7-py3-none-any.whl (3.8 kB)
Collecting Jinja2~=3.0
  Using cached Jinja2-3.1.2-py3-none-any.whl (133 kB)
Collecting wasmtime<6,>=1
  Downloading wasmtime-5.0.0-py3-none-win_amd64.whl (4.5 MB)
     ---------------------------------------- 4.5/4.5 MB 1.4 MB/s eta 0:00:00
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.2-cp310-cp310-win_amd64.whl (16 kB)
Collecting idna>=2.0
[... 140 lines cut ...]
ERROR: Assert `ofs.is_open()' failed in ../yosys-src/passes/techmap/abc9_ops.cc:950.
Traceback (most recent call last):
  File "C:\Users\Maya\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Maya\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Maya\Desktop\glasgow\venv\Scripts\glasgow.exe\__main__.py", line 7, in <module>
  File "C:\Users\Maya\Desktop\glasgow\venv\lib\site-packages\glasgow\cli.py", line 857, in main
    exit(loop.run_until_complete(_main()))
  File "C:\Users\Maya\AppData\Local\Programs\Python\Python310\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()
  File "C:\Users\Maya\Desktop\glasgow\venv\lib\site-packages\glasgow\cli.py", line 773, in _main
    f.write(plan.execute())
  File "C:\Users\Maya\Desktop\glasgow\venv\lib\site-packages\glasgow\target\hardware.py", line 118, in execute
    products  = self.lower.execute_local(build_dir)
  File "C:\Users\Maya\Desktop\glasgow\venv\lib\site-packages\amaranth\build\run.py", line 97, in execute_local
    subprocess.check_call(["cmd", "/c", "call {}.bat".format(self.script)])
  File "C:\Users\Maya\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmd', '/c', 'call build_top.bat']' returned non-zero exit status 1.
```

Running `glasgow build --rev C1 -t zip uart` produces this archive: [uart.zip](https://github.com/GlasgowEmbedded/glasgow/files/10757080/uart.zip)

A similar issue was mentioned in https://github.com/GlasgowEmbedded/glasgow/issues/298#issuecomment-1355933818, so these might be potentially related.

whitequark, 2023-02-16:
This is a YoWASP issue. YoWASP no longer mounts `/tmp` itself, and Yosys abc integration [uses that](https://libera.irclog.whitequark.org/glasgow/2023-02-16#1676559679-1676559709;) to pass intermediate data around.

whitequark, 2023-02-16:
Could you recheck using the following `yowasp-runtime` wheel, please? 
[wheel.zip](https://github.com/GlasgowEmbedded/glasgow/files/10760019/wheel.zip)

VioletEternity, 2023-02-16:
I can build a bitstream with this wheel.

whitequark, 2023-02-16:
I've released `yowasp-runtime` with this patch. Everything should work after `pip install -U yowasp-runtime` (or just reinstall the venv).

closed 2023-02-16 by whitequark