GlasgowEmbedded/glasgow #312: Documentation

attie, 2023-03-30:
I am planning to spend time writing more documentation for Glasgow, in preparation for the influx of new users via Crowd Supply.

I'm sharing my plans here, please feel free to comment with suggestions.
This will reside in the [`/docs/`](/docs/) directory, and will not likely touch [`README.md`](/README.md), at least for now... it may absorb some of the content at some point.
Some of these items will draw heavily on existing words.

The longer-term goal of this issue is to also produce static HTML that is hosted somewhere (probably using GitHub pages on a branch of this repo?)

- Index
  - Intro - what is Glasgow & what can it do?
  - Project overview
    - Revisions & naming
- Getting started
  - Installation
    - Linux
    - macOS
    - Windows
  - Running / invocation
    - Examples
  - Common problems & resolutions
- Applets
  - Overview
  - Details
  - Out-of-tree applets (_TBD..._)
  - My first applet example / template / walkthrough
- Hardware variants
  - Naming scheme
  - _foreach_
    - Feature summary
    - Board overview
- Production
  - PCBs, BOM & Assembly
  - Provisioning (serial numbers & firmware)
  - Testing
- Internals
  - Boot
  - Communication (between FPGA and Python)
  - Applet build, load & run
  - _probably much more, TBD..._
- Q&A
  - Where did the name come from?
  - Where can I get one?
  - Where can I ask for help?
  - Where & how can I use it? (Licensing)

whitequark, 2023-03-31:
Sounds wonderful! Let's gather on our IRC/Discord channel to discuss this sometime soon?

attie, 2023-09-30:
Requested examples / expansion:

- REPL usage
- Internal analyzer

attie, 2023-10-07:
- Add instructions for Ubuntu 20.04 LTS

whitequark, 2023-10-08:
> * Add instructions for Ubuntu 20.04 LTS

These are *really* annoying--I've had a go and I think I won't be working on them.