GlasgowEmbedded/glasgow #313: PIC MCU ICSP (or Enhanced ICSP) programming

matthuszagh, 2023-04-07:
This is a feature request for the ability to program PIC MCUs. Specifically, in my case, to program a PIC24FJ64GB0xx MCU. I may look into implementing this myself. But if there's been any effort on this front (I didn't see any from a quick search), please let me know. Or, if you have specific requirements for how this should be done that are not otherwise obvious, that would also be helpful.

whitequark, 2023-04-07:
@marcan looked into it, I think

whitequark, 2025-07-06:
Actually, there's another level of added confusion here: while most people understand "PIC" to refer to the 8-bit MCU series, PIC24 is essentially unrelated to that lineup. If you're still interested in working on PIC24 flashing let me know and I'll reopen this issue.

closed 2025-07-06 by whitequark