GlasgowEmbedded/glasgow #383: Instructions in README do not make it possible to build bitstreams on Windows

VioletEternity, 2023-08-03:
I followed the [instructions](https://github.com/GlasgowEmbedded/glasgow#-with-windows) exactly. After installing everything afresh, any command that involves building the bitstream causes the following error to occur:

```
>glasgow build --rev C3 uart
I: glasgow.cli: generating bitstream for applet 'uart'
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Scripts\yowasp-yosys.exe\__main__.py", line 4, in <module>
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Lib\site-packages\yowasp_yosys\__init__.py", line 2, in <module>
    import yowasp_runtime
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Lib\site-packages\yowasp_runtime\__init__.py", line 5, in <module>
    import wasmtime
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Lib\site-packages\wasmtime\__init__.py", line 16, in <module>
    from ._error import WasmtimeError, ExitTrap
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Lib\site-packages\wasmtime\_error.py", line 2, in <module>
    from . import _ffi as ffi
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Lib\site-packages\wasmtime\_ffi.py", line 26, in <module>
    raise RuntimeError("unsupported architecture for wasmtime: {}".format(machine))
RuntimeError: unsupported architecture for wasmtime:
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "c:\users\maya.maya-domain\.local\bin\glasgow.exe\__main__.py", line 7, in <module>
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\glasgow\software\glasgow\cli.py", line 901, in main
    exit(loop.run_until_complete(_main()))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\glasgow\software\glasgow\cli.py", line 798, in _main
    f.write(plan.get_bitstream())
            ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\glasgow\software\glasgow\target\hardware.py", line 164, in get_bitstream
    bitstream_data = self.execute(debug=debug)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\glasgow\software\glasgow\target\hardware.py", line 138, in execute
    products  = self.lower.execute_local(build_dir, env=self.toolchain.env_vars)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Lib\site-packages\amaranth\build\run.py", line 98, in execute_local
    subprocess.check_call(["cmd", "/c", "call {}.bat".format(self.script)],
  File "C:\Users\Maya.MAYA-DOMAIN\AppData\Local\Programs\Python\Python311\Lib\subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmd', '/c', 'call build_top.bat']' returned non-zero exit status 1.
```

The installed package versions are:

```
>pipx runpip glasgow list
Package              Version              Editable project location
-------------------- -------------------- --------------------------------------------------------
amaranth             0.4.dev160+gb952005
amaranth-yosys       0.25.0.0.post74
appdirs              1.4.4
bitarray             2.8.0
click                8.1.6
colorama             0.4.6
fx2                  0.11
glasgow              0.1.dev1786+g3146305 C:\Users\Maya.MAYA-DOMAIN\AppData\Local\glasgow\software
Jinja2               3.1.2
libusb1              3.0.0
MarkupSafe           2.1.3
packaging            23.1
pip                  23.2.1
pyvcd                0.4.0
setuptools           68.0.0
wasmtime             11.0.0
wheel                0.41.0
yowasp-nextpnr-ice40 0.6.0.0.post440
yowasp-runtime       1.30
yowasp-yosys         0.31.0.13.post553
```

Running yowasp-yosys normally succeeds:

```
>C:\Users\Maya.MAYA-DOMAIN\AppData\Local\.local\pipx\venvs\glasgow\Scripts\yowasp-yosys.exe
Preparing to run yowasp-yosys. This might take a while...
Yosys 0.31+13 (git sha1 6965abeef, ccache clang 14.0.0-1ubuntu1 -Os -flto -flto)
```

This seems to be caused by this line executing the build with only a few environment variables:

https://github.com/GlasgowEmbedded/glasgow/blob/31463053c8ca7bf9fda0c44e43f39f003765646d/software/glasgow/target/hardware.py#L138

and with these variables not including `PROCESSOR_ARCHITECTURE`:

https://github.com/GlasgowEmbedded/glasgow/blob/31463053c8ca7bf9fda0c44e43f39f003765646d/software/glasgow/target/toolchain.py#L233-L243

Wasmtime uses `platform.machine()` to determine the system architecture:

https://github.com/bytecodealliance/wasmtime-py/blob/8a759215eaf515f69a70ee04bf105c66f044fff9/wasmtime/_ffi.py#L20

This method relies on the `PROCESSOR_ARCHITECTURE` environment variable:

```
>>> subprocess.call([sys.executable, "-c", "import platform; print(platform.machine())"], env={"PROCESSOR_ARCHITECTURE": "FOO"})
FOO
```

If this environment variable is additionally included, the build succeeds and the resulting bitstream works.

attie, 2023-08-03:
Oh. Thanks for the report.

This is likely related to the "_hermetic_" builds that Amaranth is doing... see also:
- https://github.com/amaranth-lang/amaranth/commit/cb5b0e38
- https://github.com/YosysHQ/oss-cad-suite-build/issues/75

VioletEternity, 2023-08-03:
> I suspect this is related to the "_hermetic_" builds that Amaranth is doing...

As far as I can see "hermetic" builds are something that is introduced in Glasgow. Amaranth provides the necessary hooks to clear the environment, but that's it.

whitequark, 2023-08-03:
Yeah, we need to include `PROCESSOR_ARCHITECTURE` at least, on Windows.

whitequark, 2023-08-03:
@VioletEternity Can you test https://github.com/GlasgowEmbedded/glasgow/pull/384, please?

commit ba3cb99d (2023-08-04): target.hardware: whitelist `PROCESSOR_ARCHITECTURE`.

It is used by wasmtime-py to determine which DLL to load.

Fixes #383.
PR #384 (2023-08-04): target.hardware: whitelist `PROCESSOR_ARCHITECTURE`
It is used by wasmtime-py to determine which DLL to load.

Fixes #383.
closed 2023-08-04 by whitequark