GlasgowEmbedded/glasgow #386: Glasgow does not gracefully shutdown if the device is unplugged...

attie, 2023-08-05:
For example, running the `uart` applet, and then unplugging it will produce lots of output, and then hang on `E: g.cli: transfer error: LIBUSB_TRANSFER_ERROR`, requiring `^C` to return to the shell.

<details><summary>Console Output</summary>

```
user@c0f973ab2540:~$ glasgow --serial C1-20200923T174926Z run uart -V 3.3 tty
I: g.device.hardware: generating bitstream ID fc10eecac687e82738156c3ee7cdfb42
I: g.cli: running handler for applet 'uart'
I: g.applet.interface.uart: port(s) A, B voltage set to 3.3 V
I: g.applet.interface.uart: running on a TTY; enter `Ctrl+\ q` to quit
W: g.applet.interface.uart: 40 frame or parity errors detected
W: g.applet.interface.uart: 12201 frame or parity errors detected
W: g.applet.interface.uart: 12202 frame or parity errors detected
W: g.applet.interface.uart: 12214 frame or parity errors detected
W: g.applet.interface.uart: 12211 frame or parity errors detected
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-5' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error:
 LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-6' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error:
 LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-7' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error:
 LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-8' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error:
 LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-9' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error:
 LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-10' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error
: LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-11' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error
: LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-12' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error
: LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: asyncio: Task exception was never retrieved
future: <Task finished name='Task-13' coro=<DirectDemultiplexerInterface._in_task() done, defined at /opt/glasgow/software/glasgow/access/direct/demultiplexer.py:214> exception=GlasgowDeviceError('transfer error
: LIBUSB_TRANSFER_ERROR')>
Traceback (most recent call last):
  File "/opt/glasgow/software/glasgow/access/direct/demultiplexer.py", line 222, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 287, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/opt/glasgow/software/glasgow/device/hardware.py", line 257, in _do_transfer
    return await result_future
glasgow.device.GlasgowDeviceError: transfer error: LIBUSB_TRANSFER_ERROR
E: g.cli: transfer error: LIBUSB_TRANSFER_ERROR
```

</details>

whitequark, 2023-08-05:
Oh yeah. I've spent a lot of effort making it not do that and it's still an issue.

whitequark, 2023-10-21:
Fixed by https://github.com/GlasgowEmbedded/glasgow/pull/452.

whitequark, 2023-10-21:
> requiring ^C to return to the shell.

The UART applet in TTY mode specifically still requires this, but it also works with a single newline. Other applets return instantly. This could be separately fixed in the UART applet.

PR #452 (2023-10-21): Do not fill terminal with noise when a device is surprise disconnected
"Surprise disconnected" means disconnecting a device while it is being used by an applet, script, or REPL; whether there are ongoing transfers or not.

After this commit, in general, only one message should be printed:

```
E: g.cli: device disconnected
```

However, applet logic will often not handle cancellation or USB transfer errors, so certain individual applets may still print more noise.

This has been one of the more irritating papercuts, especially as all that noise will make recovering from an accidentally unplugg [... cut]
closed 2023-10-21 by attie