GlasgowEmbedded/glasgow #387: Devices may "forget" firmware.

attie, 2023-08-05:
I believe I've seen this a few times - a Glasgow with firmware flashed, will "forget" it, and require firmware to be loaded the first time `glasgow` is invoked.

- It seems that byte @ `0x0` in the EEPROM is being set to `0xC0` (i.e: "_Load custom VID / PID / DID only_").
- `fx2tool ... read_eeprom` shows that firmware was indeed present (see [gist](https://gist.github.com/attie/a9106c4f9e65fddbb815724fdb306965)).
- `glasgow flash` loads the firmware again with no issue.
- I do not currently have a way to reproduce this issue.

whitequark, 2023-08-05:
The code in the firmware that does this is here:

https://github.com/GlasgowEmbedded/glasgow/blob/ba3cb99d0b2bcfd2638ccf0d0418942ec040932f/firmware/main.c#L226-L255

It's called from `main`. But unless `glasgow_config.revision` is for some reason one of specifically `'A'`, `'B'`, or `'C'`, it won't fire.

Since this wouldn't happen with the boards @esden shipped I suspect the cause is something else.

I'm going to close this since it's completely unactionable and there is no reproducer. However, this isn't my way of saying "this issue doesn't matter"—closed issues still stay on the tracker and get searched. If anyone ever hits this again (and is able to provide more context) we will reopen it.

whitequark, 2023-08-05:
Alternatively, we can just remove `config_fixup` entirely. I think it's safe to assume that all of the devices that required the fixup have received it in the four years (git blame points at 2019-03-01) it's been in the codebase.

attie, 2023-08-05:
> Alternatively, we can just remove config_fixup entirely.

I was wondering about that approach too... or wondering if I missed something else performing the write via USB.

whitequark, 2023-08-05:
> or wondering if I missed something else performing the write via USB.

Unless you ran `glasgow flash --no-firmware` or `glasgow factory` I don't see what would issue such a write over USB...

commit 52734ce9 (2023-08-05): firmware: remove config fixup functionality.

There was only a small amount of devices that required config fixup
in first place, and they should have all been fixed in the ~4 years
that have passed since the code was added. If anyone boots firmware
on a device with non-fixed EEPROM after this commit, the device will
come up with a `Z9-9999...9` serial and running `glasgow flash` with
the right revision will fix the problem.

Fixes #387.
PR #389 (2023-08-05): firmware: remove config fixup functionality
There was only a small amount of devices that required config fixup in first place, and they should have all been fixed in the ~4 years that have passed since the code was added. If anyone boots firmware on a device with non-fixed EEPROM after this commit, the device will come up with a `Z9-9999...9` serial and running `glasgow flash` with the right revision will fix the problem.

Fixes #387.
closed 2023-08-05 by attie