GlasgowEmbedded/glasgow #400: Add host OS version to `--version` output

esden, 2023-10-02:
We ask users to provide exact OS version when reporting bugs. It would be useful to have that info included in the `glasgow --version` output to make sure the information is provided correctly and with the necessary granularity.

Alternatively we could have a separate CLI option to provide all the necessary system information to make debugging of user reported bugs easier. As it might be good to know some extra information about how the hardware is connected to the system too.

whitequark, 2023-10-02:
I have no idea how to actually implement this. Hopefully there is a library or something!

whitequark, 2023-10-02:
Not sure it's so good on Debian...

emaste, 2023-10-02:
Perhaps information from /etc/os-release? We adopted that from the Linux ecosystem, although I'm not sure if it's universal.

On my laptop it contains:
```
NAME=FreeBSD
VERSION=14.0-CURRENT
VERSION_ID=14.0
ID=freebsd
ANSI_COLOR="0;31"
PRETTY_NAME="FreeBSD 14.0-CURRENT"
CPE_NAME=cpe:/o:freebsd:freebsd:14.0
HOME_URL=https://FreeBSD.org/
BUG_REPORT_URL=https://bugs.FreeBSD.org/
```

whitequark, 2023-10-02:
> Perhaps information from /etc/os-release? We adopted that from the Linux ecosystem, although I'm not sure if it's universal.

Yep, I think that's the way to go! `/etc/os-release` on *nix and its own thing on Windows.

whitequark, 2023-10-02:
Also maybe we should add the versions of every installed package (or at least every `glasgow` dependency there) too; cf. https://github.com/GlasgowEmbedded/glasgow/issues/408, though there `--version` itself fails.

commit f3e3be34 (2023-10-10): cli: add platform, architecture, and OS version to `--version` output.

Fixes #400.
PR #440 (2023-10-10): cli: add platform, architecture, and OS version to `--version` output
Fixes #400.

Output looks like this:

```
$ pdm run glasgow --version
Glasgow 0.1.dev1826+g4fe4acc.d20231004164618 (CPython 3.9.18 on Linux-6.1.0-10-amd64-x86_64-with-glibc2.37)

$ pdm run glasgow --version
Glasgow 0.1.dev1825+gcc2acdc (CPython 3.11.2 on Linux-6.1.0-10-amd64-x86_64-with-glibc2.36 Debian GNU/Linux 12 (bookworm))

% pdm run glasgow --version
Glasgow 0.1.dev1889+gb95a16e (CPython 3.11.5 on macOS-13.4.1-arm64-arm-64bit)

>pdm run glasgow --version
Glasgow 0.1.dev1891+g3833fcb (CPython 3.11.4 on Windows-10-10.0.1 [... cut]
closed 2023-10-10 by attie