GlasgowEmbedded/glasgow #408: "glasgow --version" produces TypeError: entry_points() got an unexpected keyword argument 'group'

txf-, 2023-10-02:
OS: Windows 11 
Python: 3.9.0
importlib version : 5.1.0

I followed the installation instructions from here:
https://glasgow-embedded.org/latest/install.html

Here is the output returned when running `glasgow --version`:

```
  File "c:\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\users\tiago\.local\bin\glasgow.exe\__main__.py", line 7, in <module>
  File "C:\Users\Tiago\Documents\Git\glasgow\software\glasgow\cli.py", line 901, in main
    exit(loop.run_until_complete(_main()))
  File "c:\python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Users\Tiago\Documents\Git\glasgow\software\glasgow\cli.py", line 484, in _main
    args = get_argparser().parse_args()
  File "C:\Users\Tiago\Documents\Git\glasgow\software\glasgow\cli.py", line 277, in get_argparser
    add_applet_arg(p_run, mode="interact", required=True)
  File "C:\Users\Tiago\Documents\Git\glasgow\software\glasgow\cli.py", line 128, in add_applet_arg
    for handle, metadata in GlasgowAppletMetadata.all().items():
  File "C:\Users\Tiago\Documents\Git\glasgow\software\glasgow\support\plugin.py", line 71, in all
    return {ep.name: cls(ep) for ep in importlib.metadata.entry_points(group=cls.GROUP_NAME)}
TypeError: entry_points() got an unexpected keyword argument 'group
```
Anything wrong with the python version I'm using?

whitequark, 2023-10-02:
That's really odd! Could you run `pipx runpip glasgow freeze` and post the output please?

txf-, 2023-10-02:
Here:

```
amaranth @ git+https://github.com/amaranth-lang/amaranth.git@0c3ada62503359900f185a9b2d8f122b2d952013
amaranth-yosys==0.25.0.0.post77
appdirs==1.4.4
bitarray==2.8.1
click==8.1.7
colorama==0.4.6
fx2==0.11
-e git+https://github.com/GlasgowEmbedded/glasgow@17b22eafd48a55d5fa95f7f1fa9bddc835d30ad5#egg=glasgow&subdirectory=software
Jinja2==3.1.2
libusb1==3.0.0
MarkupSafe==2.1.3
packaging==23.2
pyvcd==0.4.0
wasmtime==13.0.2
yowasp-nextpnr-ice40==0.6.0.0.post445
yowasp-runtime==1.40
yowasp-yosys==0.33.0.0.post574
```

whitequark, 2023-10-02:
Ah, looks like our software is currently broken on 3.9.

txf-, 2023-10-02:
Which version of python do you recommend?

whitequark, 2023-10-02:
We should just fix our software to work on 3.9. Someone will do this (possibly me) whenever available.

whitequark, 2023-10-02:
I'm actually quite confused how you end up getting this error considering that we have continuous integration running on 3.9, so this isn't quite trivial to fix. It should already work! No idea why it doesn't.

attie, 2023-10-02:
> Which version of python do you recommend?

@txf- if you want to play before there is a fix, then 3.10 definately works.

emaste, 2023-10-04:
FWIW I see the same issue running glasgow on FreeBSD, w/ python 3.9.18
```
  File "/usr/home/emaste/src/glasgow/software/glasgow/support/plugin.py", line 71, in all
    return {ep.name: cls(ep) for ep in importlib.metadata.entry_points(group=cls.GROUP_NAME)}
TypeError: entry_points() got an unexpected keyword argument 'group'
```

emaste, 2023-10-04:
On https://docs.python.org/3/library/importlib.metadata.html I see:
> _Compatibility Note_
>
> The “selectable” entry points were introduced in `importlib_metadata` 3.6 and Python 3.10. Prior to those changes, `entry_points` accepted no parameters and always returned a dictionary of entry points, keyed by group.

SnoopJ, 2023-10-04:
Yep, the proximal cause here is that `importlib.metadata.entry_points()` didn't accept arguments [until Python 3.10](https://github.com/python/cpython/commit/f917efccf8d5aa2b8315d2a832a520339e668187#diff-319dbaf16f78143c0043addc0e89ee8c44594c861da9da3c160f3598d2625a35L620-R768), but Glasgow is assuming this functionality is present.

An expedient fix would be to the third-party [`importlib-metadata`](https://pypi.org/project/importlib-metadata/), or otherwise write two code paths depending on what Python version the software is running under.

PR #416 (2023-10-04): support.plugin: add python 3.8-3.9 support
Fixes #408.
closed 2023-10-04 by whitequark