GlasgowEmbedded/glasgow #418: Add the very nice diagrams from CrowdSupply packaging on the website

whitequark, 2023-10-04:
Ideally as both a printable SVG and a smaller / more focused version for on-screen reference.

victorhooi, 2024-06-09:
For reference - it should be these diagrams:

[image]
[image]

@esden Let me know if there's anything I could do to help here (this seemed like low hanging fruit for a non-electronics genius like me) - @whitequark said you had the original files. Otherwise at least the block diagram could be re-created somewhat easily using graphviz, mermaidjs, or any other vector/diagraming software?

whitequark, 2025-07-06:
@esden, could you look into this issue please?

whitequark, 2025-07-23:
I think I'll need to integrate them into the docs, which is probably easiest to do myself; I'm not yet sure exactly where I'd put them in the website, I'd have to think about it a bit.