GlasgowEmbedded/glasgow #481: Find a way to enable RAW_IO WinUSB policy when running on Windows

whitequark, 2023-10-29:
References:
- https://github.com/libusb/libusb/pull/1208
- https://learn.microsoft.com/en-us/windows-hardware/drivers/usbcon/winusb-functions-for-pipe-policy-modification

This is likely significantly limiting our latency numbers on Windows.

isabelburgos, 2024-07-03:
I'm not sure why, but I found that Windows USB performance is better with the [libusb-win32](https://github.com/mcuee/libusb-win32) driver instead of the stock WinUSB driver. I installed the libusb-win32 driver with [Zadig](https://zadig.akeo.ie).

With WinUSB:

```
I: glasgow.cli: running handler for applet 'benchmark'
I: glasgow.applet.internal.benchmark: running benchmark mode source for 8.000 MiB
E: glasgow.applet.internal.benchmark: mode source failed!
I: glasgow.applet.internal.benchmark: running benchmark mode sink for 8.000 MiB
I: glasgow.applet.internal.benchmark: mode sink: 25.59 MiB/s (204.71 Mb/s)
I: glasgow.applet.internal.benchmark: running benchmark mode loopback for 8.000 MiB
I: glasgow.applet.internal.benchmark: mode loopback: 33.04 MiB/s (264.30 Mb/s)
I: glasgow.applet.internal.benchmark: running benchmark mode latency for 8.000 MiB
I: glasgow.applet.internal.benchmark: mode latency: mean: 458.82 us stddev: 103.68 us worst: 4375.20 us
```

With libusb-win32:
```
I: glasgow.cli: running handler for applet "benchmark'
I: glasgow.applet.internal.benchmark: running benchmark mode source for 8.000 MiB
I: glasgow.applet.internal.benchmark: mode source: 42.49 MiB/s (339.91 Mb/s)
I: glasgow.applet.internal.benchmark: running benchmark mode sink for 8.000 MiB
I: glasgow.applet.internal.benchmark: mode sink: 42.49 MiB/s (339.90 Mb/s)
I: glasgow.applet.internal.benchmark: running benchmark mode loopback for 8.000 MiB
I: glasgow.applet.internal.benchmark: mode loopback: 44.33 MiB/s (354.62 Mb/s)
I: glasgow.applet.internal.benchmark: running benchmark mode latency for 8.000 MiB
I: glasgow.applet.internal.benchmark: mode latency: mean: 438.91 us stddev: 106.88 us worst: 3135.40 us
```

This was on **Windows 11**.
Specifics about this system:
- Windows 11 Home 22H2, OS build 22621.3737
- WinUSB version 10.0.22621.608
- libusb-win32 version 1.4.0.0

sonatique, 2025-08-28:
@whitequark  Can we consider this issue closed now that we have https://github.com/libusb/libusb/pull/1512 merged?

whitequark, 2025-08-28:
I think three things need to happen first:
- a libusb1 release
- a python-libusb1 release
- a glasgow patch enabling RAW_IO plus testing

whitequark, 2026-09-24:
I've discussed this with someone who has access to Microsoft source code and RAW_IO is **not** what we want here. It would cause bulk packets to be received in a random order instead of sequential.

closed 2026-09-24 by whitequark