GlasgowEmbedded/glasgow #511: JEDEC manufacturer lookup fails for bank 15

neuschaefer, 2024-01-13:
I am trying to scan a device with IDCODE=0x07926f0f, manufacturer=0x787, but `jedec_mfg_name_from_bank_num` fails:

```
$ glasgow run jtag-probe -V 3.3 --pin-tck 3 --pin-tms 2 --pin-tdi 1 --pin-tdo 4 --pin-trst 0 scan
I: g.device.hardware: device already has bitstream ID e999b72415fd3441a20e233ca2a839d9
I: g.cli: running handler for applet 'jtag-probe'
I: g.applet.interface.jtag_probe: port(s) A, B voltage set to 3.3 V
I: g.applet.interface.jtag_probe: shifted 32-bit DR=<11110000111101100100100111100000>
I: g.applet.interface.jtag_probe: shifted 4-bit IR=<1000>
I: g.applet.interface.jtag_probe: discovered 1 TAPs
Traceback (most recent call last):
  File "/.../bin/glasgow", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/.../glasgow/software/glasgow/cli.py", line 937, in main
    exit(loop.run_until_complete(_main()))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/jn/dev/fpga/glasgow/software/glasgow/cli.py", line 707, in _main
    return applet_task.result()
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/jn/dev/fpga/glasgow/software/glasgow/cli.py", line 656, in run_applet
    return await applet.interact(device, args, iface)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jn/dev/fpga/glasgow/software/glasgow/applet/interface/jtag_probe/__init__.py", line 1039, in interact
    mfg_name = jedec_mfg_name_from_bank_num(idcode.mfg_id >> 7,
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jn/dev/fpga/glasgow/software/glasgow/database/jedec.py", line 18, in jedec_mfg_name_from_bank_num
    for (number, name, code) in _manufacturers[bank]:
                                ~~~~~~~~~~~~~~^^^^^^
IndexError: list index out of range
```

The `_manufacturers` array contains bank 1 to bank 15, which amounts to 15 banks (not 16), so bank 15 is out of bounds.

I suspect that the JEP106 conversion script might have an off-by-one error and generate one too few entries.

wanda-phi, 2024-01-13:
There isn't an off by one in the code; the problem is that new manufacturer IDs have been added since the last update of the database in Glasgow, and the code deals badly with an out-of-bounds bank number (which I fix in #512).

I'd gladly update the database if you can point me to a newer version of JEP106 PDF.

wanda-phi, 2024-01-13:
Oh huh. The PDF is actually a free download. I'll just update the database then.

neuschaefer, 2024-01-13:
Thanks for the fix in #512.

The device that I'm scanning isn't particularly new (so the update doesn't help me in this case), I think it just reports a bogus manufacturer ID.

PR #512 (2024-01-13): database.jedec: fix lookup for unknown banks.
Fixes #511.
commit a2a46d22 (2024-01-13): database.jedec: fix lookup for unknown banks.

Fixes #511.
closed 2024-01-13 by wanda-phi