GlasgowEmbedded/glasgow #552: Reloading FPGA bitstream causes all outputs whose Vio is enabled to strongly drive high

whitequark, 2024-04-04:
This unfortunately happens because the reset state for the FPGA is pull-high, which translates to DIR=H (B=A), where A is FPGA-side and B is DUT-side.

Solutions:
1. Turn off Vio during FPGA reconfiguration and wait for the output cap to discharge before proceeding.
  - Pros: costs nothing.
  - Cons: slow, turns off DUT power which is undesirable, makes Vsense alert even less useful.
2. Add pulldowns on Dxn nets.
  - Pros: preserves existing design as much as feasible.
  - Cons: another BOM line, @esden has 4000 revC3 PCBs.
3. Switch to [74AXP1T45](https://github.com/GlasgowEmbedded/glasgow/issues/283) and swap A/B.
  - Pros: improves performance of the device, doesn't add a BOM line.
  - Cons: @esden has ~2000-3000 boards worth of SN74LVC1T.

Swapping A/B on SN74LVC1T isn't feasible since this level shifter is asymmetric.

smunaut, 2024-04-09:
You can also power gate the VCCIO of the FPGA bank. Either using a power switch or having two 3v3 rail ( fed by another TLV75533PDRVR with a different EN line). Not saying it's a great option, but it's a fourth option.

whitequark, 2024-04-09:
Thanks, added.

nekrondev, 2024-05-08:
Many thanks for building and maintaining `glasgow`... I've got a question regarding this issue ... how practically does this affect using the current device revision (C3)? I'm new into this and can't reason about it if it's only an issue when there is some edge case hacking where you need to power the DUT or does it interfere using `glasgow` with some common setup? As this hasn't been detected during the different PCB version integration tests I guess it should be a minor concern, right?

whitequark, 2024-05-09:
@nekrondev We'll deploy a firmware based fix that only has trivial effect on usability of device (slight delay added to firmware reload).

whitequark, 2024-06-22:
Apparently we *did* notice this... back in 2019: https://github.com/GlasgowEmbedded/glasgow/issues/96

whitequark, 2024-06-22:
While working on this (testing it with the UART applet) I discovered another buggy behavior in transients: 
[image]

Because the FIFO interface was turned off before the FPGA was reset, and because turning it off meant that several active-low flags would go permanently low, applets would receive an endless stream of phantom zero bytes. This could potentially cause very undesired operation.

closed 2024-06-22 by whitequark

u1735067, 2025-01-30:
Maybe not the right place (sorry), but to share about "*How do I verify that the fix is applied*" (https://www.crowdsupply.com/1bitsquared/glasgow/updates/campaign-fulfillment-progress): on Windows it seems the device description containing the firmware git revision is not updated after the first plugging, so the provided `Get-WmiObject` command returns the old string even after flashing, unless the device is "uninstalled" (this was deduced from https://github.com/GlasgowEmbedded/glasgow/issues/382 and tested).

whitequark, 2025-01-31:
Oh, this is good to know, thank you!

PR #588 (2024-06-22): Turn off Vio on revC before resetting FPGA as a safety measure
Fixes #552, previously reported as #96 but not fully addressed.

The minimum delay (3×t_RC for onboard bulk capacitance) is 18 ms, the actual delay implemented is 250 ms, to account for capacitance in the device under test as well.

RevAB devices are not affected.

In addition to the primary issue, another issue was discovered, where turning off the FIFO bus was causing phantom zero bytes to be received by applets, which could then exhibit undesired operation. This was addressed as well by disabling the FIFO bus after and not befor [... cut]
commit f050f518 (2024-06-22): firmware: turn off Vio on revC before resetting FPGA (-0 XRAM bytes).

Fixes #552, previously reported as #96 but not fully addressed.

The minimum delay (3×t_RC for onboard bulk capacitance) is 18 ms,
the actual delay implemented is 250 ms, to account for capacitance
in the device under test as well.

RevAB devices are not affected.

In addition to the primary issue, another issue was discovered, where
turning off the FIFO bus was causing phantom zero bytes to be received
by applets, which could then exhibit undesired operation. This was
addressed as well by disabling the FIFO bus after and n [... cut]