GlasgowEmbedded/glasgow #565: FAIL:  glasgow build --rev C3 uart

mbparks, 2024-05-04:
PS C:\Users\plebe> glasgow build --rev C3 uart
I: glasgow.cli: generating bitstream for applet 'uart'
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\plebe\pipx\venvs\glasgow\Scripts\yowasp-yosys.exe\__main__.py", line 7, in <module>
  File "C:\Users\plebe\pipx\venvs\glasgow\Lib\site-packages\yowasp_yosys\__init__.py", line 16, in _run_yosys_argv
    sys.exit(run_yosys(sys.argv[1:]))
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\pipx\venvs\glasgow\Lib\site-packages\yowasp_yosys\__init__.py", line 11, in run_yosys
    return yowasp_runtime.run_wasm(__package__, "yosys.wasm", resources=["share"],
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\pipx\venvs\glasgow\Lib\site-packages\yowasp_runtime\__init__.py", line 111, in run_wasm
    store.set_wasi(wasi_cfg)
  File "C:\Users\plebe\pipx\venvs\glasgow\Lib\site-packages\wasmtime\_store.py", line 97, in set_wasi
    raise WasmtimeError._from_ptr(error)
wasmtime._error.WasmtimeError: The system cannot find the path specified. (os error 3)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "c:\users\plebe\.local\bin\glasgow.exe\__main__.py", line 7, in <module>
  File "C:\Users\plebe\glasgow\software\glasgow\cli.py", line 937, in run_main
    exit(asyncio.run(main()))
         ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\AppData\Local\Programs\Python\Python312\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\AppData\Local\Programs\Python\Python312\Lib\asyncio\base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\Users\plebe\glasgow\software\glasgow\cli.py", line 833, in main
    f.write(plan.get_bitstream())
            ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\glasgow\software\glasgow\target\hardware.py", line 171, in get_bitstream
    bitstream_data = self.execute(debug=debug)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\glasgow\software\glasgow\target\hardware.py", line 145, in execute
    products  = self.lower.execute_local(build_dir, env=environ)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\plebe\pipx\venvs\glasgow\Lib\site-packages\amaranth\build\run.py", line 101, in execute_local
    subprocess.check_call(["cmd", "/c", f"call {self.script}.bat"],
  File "C:\Users\plebe\AppData\Local\Programs\Python\Python312\Lib\subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmd', '/c', 'call build_top.bat']' returned non-zero exit status 1.

Running Win11, and latest Python, pipx and git

whitequark, 2024-05-04:
This error is definitely a property of your environment, but it's difficult to say exactly how. The actual error most likely comes from one of [these statements](https://github.com/YoWASP/runtime-py/blob/9027ad28a03962e338dcc4392ca06260929a17ed/yowasp_runtime/__init__.py#L57-L72), although it is also possible (but unlikely) that either the temporary directory or the resource directory are missing.

Do you happen to have a `net use` or `subst` share defined that is referencing a non-existent target?

whitequark, 2024-05-04:
It looks like I just reproduced this in my environment (unintentionally, I was trying to get something done). It has a network share, but no subst mounts.

whitequark, 2024-05-04:
I've confirmed that this is an upstream (YoWASP) error and not specific to Glasgow:

[image]

whitequark, 2024-05-04:
I've confirmed that the culprit is these preopens:

```
            for letter in "abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ":
                wasi_cfg.preopen_dir(letter + ":\\", letter + ":")
```

Namely, wasmtime used to ignore invalid preopens until you accessed them, but it no longer does since version 20. The workaround for now is to downgrade `wasmtime` to version 19.0.0.

whitequark, 2024-05-04:
This is an upstream issue in the YoWASP runtime and/or Wasmtime, so I filed two issues on it:
- https://github.com/YoWASP/runtime-py/issues/3
- https://github.com/bytecodealliance/wasmtime/issues/8552

There is an automated testsuite that is supposed to prevent this from happening, however unfortunately it does not cover Windows. We could probably run it under Wine...

whitequark, 2024-05-04:
I've fixed this in https://pypi.org/manage/project/yowasp-runtime/release/1.55/ and yanked the broken release. You can upgrade the virtual environment, or just reinstall it from scratch if you don't want to bother.

lboulard, 2024-06-25:
### FAIL: `glasgow build --rev C3 uart` in presence of BitLocker drives in locked state

Follow up on same issue. This is for reference for people who get hit by same bus.
I guess that any letter without a readable file system by operating system will also trigger issue.

Hint in https://github.com/GlasgowEmbedded/glasgow/issues/565 that was used for a fix in https://github.com/YoWASP/runtime-py/issues/3 is incomplete, and current fix does not address core issue. Core issue is that drive letters on windows are dynamic and may expose ressources that are not always available as readable file system. Because not a wheel known file system — ext4, xfs, BSD partition — or, folder/drive permissions does not allow current user. Drive letters can also come and gone at any moment.

Only fix I found is altering directly pipx virtualenv for glasgow. This will ignore non accessible drive.

```patch
--- Lib\site-packages\yowasp_runtime\__init__.py.orig
+++ Lib\site-packages\yowasp_runtime\__init__.py
@@ -62,6 +62,8 @@
             for drive_index in range(26):
                 if drive_mask & (1 << drive_index):
                     drive_letter = "abcdefghijklmnopqrstuvwxyz"[drive_index]
+                    if not os.access(drive_letter + ":\\", os.R_OK):
+                        continue
                     wasi_cfg.preopen_dir(drive_letter + ":\\", drive_letter + ":")
                     drive_letter = drive_letter.upper()
                     wasi_cfg.preopen_dir(drive_letter + ":\\", drive_letter + ":")
```

Should solve most existing failure of `glasgow build` on Windows.

Currently, I understand what `__init__.py` code is doing for startup. But, I do not known why `yowasp_runtime` code does "static" capture of drives list. First fix was to only allow C drive (`drive_mask  =4`), C drive is only drive I am (almost) sure exists on any windows setup.

This is why I write bug report here until I can understand who is guilty here. And open a new issue.

I know that glasgow project is not responsable for this bug.
Still, please, reopen this bug until a final solution is found.

whitequark, 2024-06-25:
So the proper fix is to bump the YoWASP wasmtime requirement to >=21.0.0 and then bail if `wasi_cfg.preopen_dir()` fails.

whitequark, 2024-06-25:
Please test whether https://test.pypi.org/project/yowasp-runtime/1.61/ addresses the issue and if it does I will make a proper release of yowasp-runtime and bump the constraint in Glasgow.

lboulard, 2024-06-25:
@whitequark I confirm that 1.61 solved issue. Thanks.
Testes 1.62 too. working too but that was expected on Windows I think.

whitequark, 2024-06-25:
Released 1.62: https://pypi.org/project/yowasp-runtime/1.62/.

lboulard, 2024-06-25:
Confirm. Everything did run well.

```
╭─ C:\lb\Code\glasgow   main 
╰─ pipx reinstall glasgow
uninstalled glasgow! ✨ 🌟 ✨
  installed package glasgow 0.1.dev2069+gbf5f29e, installed using Python 3.11.9
  These apps are now globally available
    - glasgow.exe
done! ✨ 🌟 ✨

╭─ C:\lb\Code\glasgow   main 
╰─ pipx runpip glasgow list
Package              Version
-------------------- --------------------
amaranth             0.5.0
amaranth-yosys       0.40.0.0.post96
click                8.1.7
colorama             0.4.6
fx2                  0.13
glasgow              0.1.dev2069+gbf5f29e
importlib_resources  6.4.0
Jinja2               3.1.4
jschon               0.11.1
libusb1              3.1.0
MarkupSafe           2.1.5
packaging            24.1
pip                  24.1
platformdirs         4.2.2
pyvcd                0.4.0
rfc3986              2.0.0
setuptools           69.0.2
typing_extensions    4.12.2
wasmtime             22.0.0
wheel                0.42.0
yowasp-nextpnr-ice40 0.7.0.0.post519
yowasp-runtime       1.62
yowasp-yosys         0.42.0.0.post734

╭─ C:\lb\Code\glasgow   main 
╰─ glasgow build --rev C3 uart
I: glasgow.cli: generating bitstream for applet 'uart'
```

closed 2024-06-25 by whitequark