GlasgowEmbedded/glasgow #577: Installation issue on macOS

mringwal, 2024-06-02:
Hi. eagerly awaiting my Glasgow, I've followed the installation instructions for macOS and get an error checking the version. 
After installing pipx with brew, re-starting the terminal, I get this:

```
➜  Projects pipx reinstall glasgow
uninstalled glasgow! ✨ 🌟 ✨
  installed package glasgow 0.1.dev1654+gb1985b3, installed using Python 3.12.3
  These apps are now globally available
    - glasgow
done! ✨ 🌟 ✨
➜  Projects glasgow --version
Traceback (most recent call last):
  File "/Users/mringwal/.local/bin/glasgow", line 5, in <module>
    from glasgow.cli import main
  File "/Users/mringwal/Projects/glasgow/software/glasgow/cli.py", line 31, in <module>
    from .applet import *
  File "/Users/mringwal/Projects/glasgow/software/glasgow/applet/__init__.py", line 108, in <module>
    from ..access.simulation import *
  File "/Users/mringwal/Projects/glasgow/software/glasgow/access/simulation/__init__.py", line 3, in <module>
    from .demultiplexer import *
  File "/Users/mringwal/Projects/glasgow/software/glasgow/access/simulation/demultiplexer.py", line 13, in <module>
    class SimulationDemultiplexerInterface(AccessDemultiplexerInterface):
  File "/Users/mringwal/Projects/glasgow/software/glasgow/access/simulation/demultiplexer.py", line 20, in SimulationDemultiplexerInterface
    @asyncio.coroutine
     ^^^^^^^^^^^^^^^^^
AttributeError: module 'asyncio' has no attribute 'coroutine'. Did you mean: 'coroutines'?```

Any ideas/suggestion?

whitequark, 2024-06-02:
> installed package glasgow 0.1.dev1654+gb1985b3, installed using Python 3.12.3

That looks like an extremely old checkout (~400 commits behind). You should `git pull`.

mringwal, 2024-06-02:
You're correct. I did a git pull but have been on the old-school 'master' branch. After switching to 'main', it starts as expected. Thanks!

closed 2024-06-02 by mringwal