GlasgowEmbedded/glasgow #599: Migrate away from deprecated `Pads`/`get_pads` mechanism

whitequark, 2024-06-26:
The two mechanisms are similar conceptually and mainly differ syntactically. The following command can illustrate the changes required:

```console
$ git diff 38730032..50b9ea58 -- glasgow/applet/interface/jtag_probe/__init__.py
```

List of applets that still need migration:

- [ ] applet.audio.dac
- [ ] applet.audio.yamaha_opx
- [ ] applet.control.servo
- [ ] applet.display.hd44780
- [ ] applet.display.pdi
- [ ] applet.interface.analyzer
- [ ] applet.interface.i2c_initiator
- [ ] applet.interface.i2c_target
- [ ] applet.interface.ps2_host
- [ ] applet.interface.sbw_probe
- [ ] applet.interface.spi_controller
- [ ] applet.interface.swd_openocd
- [ ] applet.interface.uart
- [ ] applet.memory.floppy
- [ ] applet.memory.onfi
- [ ] applet.memory.prom
- [ ] applet.program.avr.spi
- [ ] applet.program.m16c
- [ ] applet.program.nrf24lx1
- [ ] applet.radio.nrf24l01
- [ ] applet.sensor.hx711
- [ ] applet.sensor.pmsx003
- [ ] applet.video.hub75_output
- [ ] applet.video.rgb_input
- [ ] applet.video.vga_output

q3k, 2024-06-26:
Like this? (#602) My Amaranth is very rusty and I don't have how to actually test this. Is there an easy way to make sure an Amaranth/Python change is an RTL no-op?

whitequark, 2024-06-26:
> Is there an easy way to make sure an Amaranth/Python change is an RTL no-op?

No, but the changes are simple enough I expect to catch most issues during review, even if it's tedious.

Improvements to the general applet infrastructure will continue at a decent pace for the next half year or so, and I expect I'll have to sit down and manually test every single applet at the end of it, hopefully while also adding some better tests. It's going to be painful but what can you do?..

whitequark, 2024-06-26:
Also I don't even know where *my* HX711 board is, much less have a load cell on hand...

q3k, 2024-06-26:
Any tips on sequencing updates across multiple modules, eg. with UART(pads/ports) being used by multiple applets?

Is there a temporary/transitional mechanism that I could use to first update the applets (or libraries?) and then the other side?

whitequark, 2024-06-26:
I'd suggest leaving updating the libraries (UART and I2C in particular) to some later point, they're kind of a pain and I'm going to rewrite UART at least soon, anyway.

For the rest, unfortunately we don't have real applet composability so the answer is suffering. Making things composable is on my TODO list but it goes after this.

isabelburgos, 2024-08-05:
One more item for this list - `examples.boilerplate` still uses pads

PR #790 (2025-04-27): Remove deprecated way to instantiate pin buffers
Closes #599.

Thanks @isabelburgos and @q3k!
commit b404cd1e (2025-04-27): software: remove deprecated way to instantiate pin buffers.

Closes #599.

Thanks @isabelburgos and @q3k!
closed 2025-04-27 by whitequark