GlasgowEmbedded/glasgow #608: [Feature request] signal generator applet

ikarus23, 2024-06-27:
Hi. It would be cool to have simple signal generator applet. I'm not sure about DACs on the Glasgow, but even a square signal from a normal IO would be great.

I found the `control-servo` applet which generates a limited PWM signal. This is a great start, but it would be great to have more freedom. Some features I'm thinking about:

1. Simple square signal defined by providing a pulse with, a pause with and a pin.
2. "Replaying" a signal: same as 1. but a string of 1 and 0 can be provided with 1 being a high signal using the pulse with and a 0 being a low signal using the pause width.
3. Same as 1. or 2. but make it parallel. This would allow to specify multiple pins and to simulate a parallel bus signal.

This would also be great for some quick and rudimentary bit banging tests.

whitequark, 2024-06-27:
My concern here is the degree of generality. Ultimately, you could consider *any* applet a type of "signal generator" since that's what Glasgow does. I think I can't accept such an applet upstream without a clearly defined and justified scope.

whitequark, 2024-06-27:
Ultimately I also want it to be easy enough to add new applets that you wouldn't have to configure some monster with dozens of features to generate a signal similar to what you want; you would write a little bit of gateware that is exactly what you want.

arnobert, 2024-11-26:
> Hi. It would be cool to have simple signal generator applet. I'm not sure about DACs on the Glasgow, but even a square signal from a normal IO would be great.
>
> I found the `control-servo` applet [...]

A simple sig gen is one of the *basic* features, I would expect. Just dial in frequency and amplitude, maybe waveform (sine, square, triangle), that's it.

whitequark, 2024-11-26:
Glasgow can't generate sine or triangle without an external LPF.

arnobert, 2024-11-27:
> Glasgow can't generate sine or triangle without an external LPF.

Even a square gen would actually help me.

rabagliati, 2024-11-27:
I also have an Analog Discovery 2, which I use a lot. I mostly use it as a digital signal analyser and protocol decoder, but I like the 'scope functionality, and the LCR addon is useful, though the dedicated ones are easier. I do not use the signal generator very much, but it is used for the LCR addon.

The Glasgow has more output capability, and its single channel analog input has a much more useful voltage range than the Analog discovery 2.

I think a signal generator and single channel scope are natural targets for the Glasgow - even if they are low frequency.

whitequark, 2024-11-27:
> I think a signal generator and single channel scope are natural targets for the Glasgow - even if they are low frequency.

An arbitrary waveform generator would need an analog frontend, and this is actually already covered: `audio-dac` plus a filter or a more complex frontend gives you that capability.

A binary pulse train generator can get fairly complex if you want multi-channel synchronization, on-the-fly updates, or things like that (you could write a PhD on this topic alone) so without a well defined scope I don't want to have one in-tree.

The analog input has something like 1 kHz of bandwidth, which is more of a "multimeter" range than a "scope" range. I don't think we currently have infrastructure to utilize even that at full capacity, and I'm not sure that infrastructure would be easy to fit into the FX2 either.

arnobert, 2024-11-27:
> > Even a square gen would actually help me.
> 
> Something scoped to a square gen would work. How do you imagine controlling it? Do you want to change the signal parameters on the fly? Do you want cycle-accurate event scheduling or anything like that?

I would try to keep it as simple as possible: Just one single channel, frequency and amplitude as parameters, without any changes during runtime, that's it.

whitequark, 2024-11-27:
Glasgow does not have per-pin amplitude control. (You can control it indirectly via the port voltage, but our command line option system isn't really currently flexible enough to express the parameters as "frequency plus amplitude", although arguably it should be.)

Other than that sounds fine.

whitequark, 2024-11-27:
> As long as there is only a single channel in use, that should be fine.

A single channel per port (you can have two!). To be clear, I never wanted to imply this couldn't work electrically, this is purely a user interface limitation.

mayjack0312, 2025-07-05:
> This code will not output a 1 kHz square wave. Did "AI" write it for you?

Ah, that's right. I only know it is a square wave (0 and 1) I use this program to make a running light example

mayjack0312, 2025-07-05:
> [@mayjack0312](https://github.com/mayjack0312) I would recommend using the [script mode](https://glasgow-embedded.org/latest/use/repl-script.html), which will let you do "PWM" like this at maybe 100 Hz (but not reliably). In order to reliably output a square wave at 1 kHz (or faster) you will need to write some gateware.

OK, I will study the document. Thank you

mayjack0312, 2025-07-05:
> [@mayjack0312](https://github.com/mayjack0312) Once you try that out, I recommend copying `examples/boilerplate.py` into a new applet and making it do PWM properly. Here is the gateware you can use as a starting point:
> 
> class PulseComponent(wiring.Component):
>     period: In(16)
> 
>     def __init__(self, ports):
>         self._ports = ports
> 
>         super().__init__()
> 
>     def elaborate(self, platform):
>         m = Module()
> 
>         m.submodules.out_buffer = out_buffer = io.Buffer("o",  self._ports.out)
> 
>         count = Signal.like(period)
>         with m.If(count == 0):
>             m.d.sync += out_buffer.o.eq(~out_buffer.o)
>             m.d.sync += count.eq(self.period)
>         with m.Else():
>             m.d.sync += count.eq(count - 1)
> 
>         return m

The test results are as follows
```

```

The following table can intuitively express the frequency bottleneck

| Target Frequency (Hz) | Target Period (ms) | High Time (ms) | Low Time (ms) | Actual Frequency (Hz) | Error (%) | Test Duration (s) | Generated Cycles |
|----------------------|-------------------|----------------|---------------|----------------------|-----------|-------------------|------------------|
| 10                   | 100.000           | 50.000         | 50.000        | 9.5                  | 5.1       | 3.06              | 29               |
| 15                   | 66.667            | 33.333         | 33.333        | 13.7                 | 8.5       | 3.06              | 42               |
| 20                   | 50.000            | 25.000         | 25.000        | 18.1                 | 9.7       | 3.04              | 55               |
| 25                   | 40.000            | 20.000         | 20.000        | 22.0                 | 11.9      | 3.04              | 67               |
| 30                   | 33.333            | 16.667         | 16.667        | 25.7                 | 14.4      | 3.04              | 78               |
| 35                   | 28.571            | 14.286         | 14.286        | 28.5                 | 18.7      | 3.02              | 86               |
| 40                   | 25.000            | 12.500         | 12.500        | 32.7                 | 18.2      | 3.03              | 99               |
| 45                   | 22.222            | 11.111         | 11.111        | 35.0                 | 22.2      | 3.03              | 106              |
| 50                   | 20.000            | 10.000         | 10.000        | 40.0                 | 19.9      | 3.02              | 121              |
| 55                   | 18.182            | 9.091          | 9.091         | 41.1                 | 25.4      | 3.02              | 124              |
| 60                   | 16.667            | 8.333          | 8.333         | 45.0                 | 24.9      | 3.02              | 136              |
| 65                   | 15.385            | 7.692          | 7.692         | 48.9                 | 24.8      | 3.01              | 147              |
| 70                   | 14.286            | 7.143          | 7.143         | 49.8                 | 28.8      | 3.01              | 150              |
| 75                   | 13.333            | 6.667          | 6.667         | 54.8                 | 27.0      | 3.01              | 165              |
| 80                   | 12.500            | 6.250          | 6.250         | 55.8                 | 30.3      | 3.01              | 168              |
| 85                   | 11.765            | 5.882          | 5.882         | 60.5                 | 28.9      | 3.01              | 182              |
| 90                   | 11.111            | 5.556          | 5.556         | 62.9                 | 30.1      | 3.00              | 189              |
| 95                   | 10.526            | 5.263          | 5.263         | 62.8                 | 33.9      | 3.01              | 189              |
| 100                  | 10.000            | 5.000          | 5.000         | 68.6                 | 31.4      | 3.00              | 206              |

Its optimal working range: 10-25Hz, the error is controlled within 12%. I think its stable upper limit is 70Hz.

The highest frequency it can generate is 173.6Hz, but the error is a bit too high.

mayjack0312, 2025-07-05:
> Yes, the GPIO method is very limited. You need to use the gateware method to achieve good results. With it, you can generate frequencies up to 24 MHz, although the error gets higher the higher you go.

"Just use it" It is mainly used as a universal burner ^_^

whitequark, 2025-07-05:
What's ProtoBlade?

mayjack0312, 2025-07-05:
> What's ProtoBlade?

Oh, I changed the name of another device O(∩_∩)O ~