GlasgowEmbedded/glasgow #667: Off-by-one in `glasgow.gateware.clockgen.ClockGen`

mbikovitsky, 2024-08-15:
The `ClockGen` module has an off-by-one error when deriving a clock using `.derive` or `.calculate`. The output clock period is one input clock cycle shorter than necessary. This happens only when the ratio between the input and output frequencies is >= 3.

[Amaranth playground link](https://amaranth-lang.org/play/#eyJhdiI6InYwLjUuMSIsInMiOiJmcm9tIGFtYXJhbnRoIGltcG9ydCAqXG5mcm9tIGFtYXJhbnRoLnNpbSBpbXBvcnQgU2ltdWxhdG9yXG5mcm9tIGFtYXJhbnRoLmJhY2sgaW1wb3J0IHJ0bGlsLCB2ZXJpbG9nXG5pbXBvcnQgYW1hcmFudGhfcGxheWdyb3VuZFxuXG5cbmNsYXNzIENsb2NrR2VuKEVsYWJvcmF0YWJsZSk6XG4gICAgXCJcIlwiXG4gICAgQSBjbG9jayBnZW5lcmF0b3IuIFRoZSBwdXJwb3NlIG9mIGEgY2xvY2sgZ2VuZXJhdG9yIGlzIHRvIHVzZSBhbiBpbnB1dCBjbG9jayBzaWduYWwgdG9cbiAgICBnZW5lcmF0ZSBhbiBvdXRwdXQgY2xvY2sgKDUwJSBkdXR5IGN5Y2xlIHB1bHNlcykgYW5kIHJpc2luZy9mYWxsaW5nIHN0cm9iZSAoMSBpbnB1dCBjbG9ja1xuICAgIHBlcmlvZCB3aWRlIHB1bHNlcyBwcmVjZWRpbmcgdGhlIHJpc2luZy9mYWxsaW5nIGVkZ2Ugb2Ygb3V0cHV0IGNsb2NrIHB1bHNlcykgc2lnbmFscyB1c2luZ1xuICAgIG5vdGhpbmcgYnV0IExVVC9GRiBsb2dpYywgaS5lLiBubyBQTExzLCBoYXJkIGRpdmlkZXJzLCBldGMuIFRoaXMgaW1wbGllcyB0aGF0IHRoZSBvdXRwdXQgY2xvY2tcbiAgICBoYXMgbm8gcHJlY2lzZSBwaGFzZSByZWxhdGlvbnNoaXAgdG8gdGhlIGlucHV0IGNsb2NrLlxuXG4gICAgVGhlcmUgYXJlIHR3byBwcmltYXJ5IGNvbnRleHRzIHdoZXJlIHRoaXMgaXMgdXNlZnVsLiBUaGUgZmlyc3QgaXMgdXNpbmcgdGhlIHN5c3RlbSBjbG9ja1xuICAgIHRvIGdlbmVyYXRlIGEgZGVyaXZlZCBjbG9ja3MgZm9yIHNvbWUgc3luY2hyb25vdXMgaW50ZXJmYWNlLCB3aGljaCBkb2VzIG5vdCBoYXZlIHRvIGJlIHZlcnlcbiAgICBwcmVjaXNlIG9yIGluIHBoYXNlIHRvIGFueSBvdGhlciBjbG9jaywgYnV0IGhhcyB0byBjb3ZlciBhIHZlcnkgd2lkZSByYW5nZSBvZiBmcmVxdWVuY2llcy5cbiAgICBGb3IgZXhhbXBsZSwgYW4gU1BJIGludGVyZmFjZSB3aXRoIGEgdXNlci1kZWZpbmVkIGZyZXF1ZW5jeSBzaG91bGQgY292ZXIgdGhlIHJhbmdlIGZyb20sXG4gICAgYXQgbGVhc3QsIDIwIE1IeiB0byAxIGtIei0tdGhhdCdzIGZpdmUgb3JkZXJzIG9mIG1hZ25pdHVkZS4gQSBQTEwgaXMgZmxleGlibGUsIGJ1dCB3b3VsZFxuICAgIG5vdCBiZSBhYmxlIHRvIGNvdmVyIG1vc3Qgb2YgdGhpcyByYW5nZSwgYW5kIGl0IGlzIGFsc28gYSB2ZXJ5IHNjYXJjZSByZXNvdXJjZTsgYSBjb3VudGVyXG4gICAgaXMgY2hlYXAgYW5kIGxpbWl0ZWQgb25seSBpbiByZXNvbHV0aW9uLlxuXG4gICAgVGhlIHNlY29uZCBpcyB0cmlnZ2VyaW5nIGV2ZW50cyBpbiB0aGUgc3lzdGVtIGNsb2NrIGRvbWFpbiBhdCBhbiBhcHByb3hpbWF0ZSBwaGFzZSByZWxhdGlvbnNoaXBcbiAgICB0byB0aGUgZGVyaXZlZCBjbG9jayBkb21haW4sIHR5cGljYWxseSB0byBhKSBzZXR1cCBvciBzYW1wbGUgYSBidXMsIGFuZCBiKSBpbnNlcnQgd2FpdCBjeWNsZXMuXG4gICAgVGhlIG91dHB1dCBvZiBhIFBMTCBjYW4gYmUgdXNlZCB0byBkbyBzbywgYnV0IGl0IGNvbXBsaWNhdGVzIHRpbWluZzsgYSBzdHJvYmUgc2lnbmFsIGZ1bGx5XG4gICAgc3luY2hyb25vdXMgdG8gdGhlIHN5c3RlbSBjbG9jayBkb21haW4gaXMgZ3VhcmFudGVlZCB0byB3b3JrLlxuXG4gICAgVGhlIHByaW1hcnkgbGltaXRhdGlvbnMgb2YgdGhpcyBhcHByb2FjaCBpcyB0aGF0IHRoZSBvdXRwdXQgZnJlcXVlbmN5IGlzIGF0IG1vc3QgdGhlIHNhbWUgYXNcbiAgICB0aGUgaW5wdXQgZnJlcXVlbmN5LCBhbmQgdGhlIG1heGltdW0gZXJyb3IsIGkuZS4gZGlzY3JlcGFuY3kgYmV0d2VlbiByZXF1ZXN0ZWQgYW5kIGFjdHVhbFxuICAgIG91dHB1dCBmcmVxdWVuY3ksIGNhbiBiZSBhcyBoaWdoIGFzIDMzLigzKSUsIGdlbmVyYWxseSBpbmNyZWFzaW5nIHRvd2FyZHMgdGhlIGhpZ2hlciBlbmQgb2ZcbiAgICB0aGUgb3V0cHV0IGZyZXF1ZW5jeSByYW5nZS5cblxuICAgIFdoaWxlIHRoZSBnYXRld2FyZSBtYXkgYXBwZWFyIHRyaXZpYWwgKGEgY291bnRlciBhbmQgYSByZWdpc3RlciksIHRoZXJlIGFyZSB0d28gZWRnZSBjYXNlc1xuICAgIHRoYXQgbmVlZCB0byBiZSBoYW5kbGVkLCBuYW1lbHkgb3V0cHV0IG9mIDUwJSBvZiBpbnB1dCBmcmVxdWVuY3kgYW5kIDEwMCUgb2YgaW5wdXQgZnJlcXVlbmN5LlxuXG4gICAgOnR5cGUgY3ljOiBpbnRcbiAgICA6cGFyYW0gY3ljOlxuICAgICAgICBPdXRwdXQgY2xvY2sgcGVyaW9kLCBpbiB0ZXJtcyBvZiBpbnB1dCBjbG9jayBwZXJpb2RzLiBVc2UgOm1ldGg6YGRlcml2ZWAgdG8gY29tcHV0ZSB0aGlzXG4gICAgICAgIHZhbHVlLCBjaGVjayBmb3IgZWRnZSBjYXNlcywgYW5kIChvcHRpb25hbGx5KSBsb2cgdGhlIGRldmlhdGlvbiBmcm9tIHJlcXVlc3RlZCBmcmVxdWVuY3lcbiAgICAgICAgYXMgd2VsbCBhcyA1MCUgZHV0eSBjeWNsZS5cbiAgICBcIlwiXCJcblxuICAgIGRlZiBfX2luaXRfXyhzZWxmLCBjeWMpOlxuICAgICAgICBzZWxmLmN5YyA9IGN5Y1xuXG4gICAgICAgIHNlbGYuY2xrICAgPSBTaWduYWwoKVxuICAgICAgICBzZWxmLnN0Yl9yID0gU2lnbmFsKClcbiAgICAgICAgc2VsZi5zdGJfZiA9IFNpZ25hbCgpXG5cbiAgICBkZWYgZWxhYm9yYXRlKHNlbGYsIHBsYXRmb3JtKTpcbiAgICAgICAgbSA9IE1vZHVsZSgpXG5cbiAgICAgICAgaWYgc2VsZi5jeWMgPT0gMDpcbiAgICAgICAgICAgICMgU3BlY2lhbCBjYXNlOiBvdXRwdXQgZnJlcXVlbmN5IGVxdWFsIHRvIGlucHV0IGZyZXF1ZW5jeS5cbiAgICAgICAgICAgICMgSW1wbGVtZW50YXRpb246IHdpcmUuXG4gICAgICAgICAgICBtLmQuY29tYiArPSBbXG4gICAgICAgICAgICAgICAgc2VsZi5jbGsuZXEoQ2xvY2tTaWduYWwoKSksXG4gICAgICAgICAgICAgICAgc2VsZi5zdGJfci5lcSgxKSxcbiAgICAgICAgICAgICAgICBzZWxmLnN0Yl9mLmVxKDEpLFxuICAgICAgICAgICAgXVxuXG4gICAgICAgIGlmIHNlbGYuY3ljID09IDE6XG4gICAgICAgICAgICAjIFNwZWNpYWwgY2FzZTogb3V0cHV0IGZyZXF1ZW5jeSBoYWxmIG9mIGlucHV0IGZyZXF1ZW5jeS5cbiAgICAgICAgICAgICMgSW1wbGVtZW50YXRpb246IGZsaXAtZmxvcC5cbiAgICAgICAgICAgIG0uZC5zeW5jICs9IFtcbiAgICAgICAgICAgICAgICBzZWxmLmNsay5lcSh+c2VsZi5jbGspLFxuICAgICAgICAgICAgXVxuICAgICAgICAgICAgbS5kLmNvbWIgKz0gW1xuICAgICAgICAgICAgICAgIHNlbGYuc3RiX3IuZXEofnNlbGYuY2xrKSxcbiAgICAgICAgICAgICAgICBzZWxmLnN0Yl9mLmVxKHNlbGYuY2xrKSxcbiAgICAgICAgICAgIF1cblxuICAgICAgICBpZiBzZWxmLmN5YyA+PSAyOlxuICAgICAgICAgICAgIyBHZW5lcmFsIGNhc2UuXG4gICAgICAgICAgICAjIEltcGxlbWVudGF0aW9uOiBjb3VudGVyLlxuICAgICAgICAgICAgY291bnRlciA9IFNpZ25hbChyYW5nZShzZWxmLmN5YykpXG4gICAgICAgICAgICBjbGtfciAgID0gU2lnbmFsKClcblxuICAgICAgICAgICAgbS5kLnN5bmMgKz0gY291bnRlci5lcShjb3VudGVyIC0gMSlcbiAgICAgICAgICAgIHdpdGggbS5JZihjb3VudGVyID09IDApOlxuICAgICAgICAgICAgICAgIG0uZC5zeW5jICs9IGNvdW50ZXIuZXEoc2VsZi5jeWMgLSAxKVxuXG4gICAgICAgICAgICB3aXRoIG0uSWYoY291bnRlciA9PSBzZWxmLmN5YyAvLyAyKTpcbiAgICAgICAgICAgICAgICBtLmQuc3luYyArPSBzZWxmLmNsay5lcSgxKVxuICAgICAgICAgICAgd2l0aCBtLklmKGNvdW50ZXIgPT0gMCk6XG4gICAgICAgICAgICAgICAgbS5kLnN5bmMgKz0gc2VsZi5jbGsuZXEoMClcblxuICAgICAgICAgICAgbS5kLnN5bmMgKz0gY2xrX3IuZXEoc2VsZi5jbGspXG4gICAgICAgICAgICBtLmQuY29tYiArPSBbXG4gICAgICAgICAgICAgICAgc2VsZi5zdGJfci5lcSh+Y2xrX3IgJiAgc2VsZi5jbGspLFxuICAgICAgICAgICAgICAgIHNlbGYuc3RiX2YuZXEoIGNsa19yICYgfnNlbGYuY2xrKSxcbiAgICAgICAgICAgIF1cblxuICAgICAgICByZXR1cm4gbVxuXG4gICAgQHN0YXRpY21ldGhvZFxuICAgIGRlZiBjYWxjdWxhdGUoaW5wdXRfaHosIG91dHB1dF9oeiwgbWF4X2RldmlhdGlvbl9wcG09Tm9uZSwgbWluX2N5Yz1Ob25lKTpcbiAgICAgICAgXCJcIlwiXG4gICAgICAgIENhbGN1bGF0ZSB0aGUgaW50ZWdlciBwZXJpb2QgcmF0aW8gZm9yIGRpdmlkaW5nIGFuIGBgaW5wdXRfaHpgYCBjbG9jayB0byBhbiBhcHByb3hpbWF0ZWx5XG4gICAgICAgIGBgb3V0cHV0X2h6YGAgY2xvY2ssIGFuZCByZXR1cm4gdGhlIGRpdmlzb3IgYXMgd2VsbCBhcyB0aGUgYWN0dWFsIG91dHB1dCBmcmVxdWVuY3kgYW5kXG4gICAgICAgIGl0cyBkZXZpYXRpb24gZnJvbSByZXF1ZXN0ZWQgb3V0cHV0IGZyZXF1ZW5jeS5cblxuICAgICAgICBSYWlzZXMgYGBWYWx1ZUVycm9yYGAgb24gYW55IG9mIHRoZSBmb2xsb3dpbmcgY29uZGl0aW9uczpcblxuICAgICAgICAgICAgKiBUaGUgb3V0cHV0IGZyZXF1ZW5jeSBpcyBoaWdoZXIgdGhhbiBpbnB1dCBmcmVxdWVuY3kuXG4gICAgICAgICAgICAqIFRoZSBvdXRwdXQgZnJlcXVlbmN5IGRpZmZlcnMgZnJvbSByZXF1ZXN0ZWQgb3V0cHV0IGZyZXF1ZW5jeSBieSBtb3JlIHRoYW5cbiAgICAgICAgICAgICAgYGBtYXhfZGV2aWF0aW9uX3BwbWBgIHBhcnRzIHBlciBtaWxsaW9uLlxuICAgICAgICAgICAgKiBUaGUgb3V0cHV0IHBlcmlvZCBpcyBsb3dlciB0aGFuIGBgbWluX2N5Y2BgIGlucHV0IHBlcmlvZHMuXG4gICAgICAgIFwiXCJcIlxuICAgICAgICBpZiBvdXRwdXRfaHogPD0gMDpcbiAgICAgICAgICAgIHJhaXNlIFZhbHVlRXJyb3IoXCJvdXRwdXQgZnJlcXVlbmN5IHs6LjNmfSBrSHogaXMgbm90IHBvc2l0aXZlXCJcbiAgICAgICAgICAgICAgICAgICAgICAgICAgICAgLmZvcm1hdChvdXRwdXRfaHogLyAxMDAwKSlcbiAgICAgICAgaWYgb3V0cHV0X2h6ID4gaW5wdXRfaHo6XG4gICAgICAgICAgICByYWlzZSBWYWx1ZUVycm9yKFwib3V0cHV0IGZyZXF1ZW5jeSB7Oi4zZn0ga0h6IGlzIGhpZ2hlciB0aGFuIGlucHV0IGZyZXF1ZW5jeSBcIlxuICAgICAgICAgICAgICAgICAgICAgICAgICAgICBcIns6LjNmfSBrSHpcIlxuICAgICAgICAgICAgICAgICAgICAgICAgICAgICAuZm9ybWF0KG91dHB1dF9oeiAvIDEwMDAsIGlucHV0X2h6IC8gMTAwMCkpXG4gICAgICAgIGlmIG1pbl9jeWMgaXMgbm90IE5vbmUgYW5kIG91dHB1dF9oeiAqIG1pbl9jeWMgPiBpbnB1dF9oejpcbiAgICAgICAgICAgIHJhaXNlIFZhbHVlRXJyb3IoXCJvdXRwdXQgZnJlcXVlbmN5IHs6LjNmfSBrSHogcmVxdWlyZXMgYSBwZXJpb2Qgc21hbGxlciB0aGFuIHs6ZH0gXCJcbiAgICAgICAgICAgICAgICAgICAgICAgICAgICAgXCJjeWNsZXMgYXQgaW5wdXQgZnJlcXVlbmN5IHs6LjNmfSBrSHpcIlxuICAgICAgICAgICAgICAgICAgICAgICAgICAgICAuZm9ybWF0KG91dHB1dF9oeiAvIDEwMDAsIG1pbl9jeWMsIGlucHV0X2h6IC8gMTAwMCkpXG5cbiAgICAgICAgY3ljID0gcm91bmQoaW5wdXRfaHogLy8gb3V0cHV0X2h6KSAtIDFcbiAgICAgICAgYWN0dWFsX291dHB1dF9oeiA9IGlucHV0X2h6IC8gKGN5YyArIDEpXG4gICAgICAgIGRldmlhdGlvbl9wcG0gPSByb3VuZCgxMDAwMDAwICogKGFjdHVhbF9vdXRwdXRfaHogLSBvdXRwdXRfaHopIC8vIG91dHB1dF9oeilcblxuICAgICAgICBpZiBtYXhfZGV2aWF0aW9uX3BwbSBpcyBub3QgTm9uZSBhbmQgZGV2aWF0aW9uX3BwbSA+IG1heF9kZXZpYXRpb25fcHBtOlxuICAgICAgICAgICAgcmFpc2UgVmFsdWVFcnJvcihcIm91dHB1dCBmcmVxdWVuY3kgezouM2Z9IGtIeiBkZXZpYXRlcyBmcm9tIHJlcXVlc3RlZCBmcmVxdWVuY3kgXCJcbiAgICAgICAgICAgICAgICAgICAgICAgICAgICAgXCJ7Oi4zZn0ga0h6IGJ5IHs6ZH0gcHBtLCB3aGljaCBpcyBoaWdoZXIgdGhhbiB7OmR9IHBwbVwiXG4gICAgICAgICAgICAgICAgICAgICAgICAgICAgIC5mb3JtYXQoYWN0dWFsX291dHB1dF9oeiAvIDEwMDAsIG91dHB1dF9oeiAvIDEwMDAsXG4gICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgICAgZGV2aWF0aW9uX3BwbSwgbWF4X2RldmlhdGlvbl9wcG0pKVxuXG4gICAgICAgIHJldHVybiBjeWMsIGFjdHVhbF9vdXRwdXRfaHosIGRldmlhdGlvbl9wcG1cblxuICAgIEB [... cut ...]
On a related note, the special case where the output frequency is half the input frequency has a strobe behaviour different from the normal case. In the special case `stb_r` and `stb_f` rise one cycle _before_ their respective clock edges. In the normal case they rise _on the same cycle_. Not sure which behaviour is the correct one, but I think they should at least be consistent.

In terms of actual impact the whole thing may not be a big deal. Assuming most applets derive clocks that are much slower than the system clock, the off-by-one deviation should be negligible.

whitequark, 2024-08-16:
Thanks! The whole idea with `ClockGen` was a little misguided (I can say as the person who introduced it, four years later), so I'm not sure if it even should be used after all. The QSPI controller applet shows what I would consider the best current practice.

whitequark, 2024-08-16:
I think one big concern here is that a bunch of applets use `ClockGen.derive` without using `ClockGen` itself. In that case fixing an off-by-1 here could introduce an off-by-1 there, which is why I hesitate to just apply the fix.

mbikovitsky, 2024-08-16:
I could search the codebase for any uses of `derive` that don't pass it to `ClockGen` and `+ 1` them to keep the original behaviour :) It's not exactly pretty, and if you say `ClockGen` shouldn't be used at all then maybe it's better to just leave it as-is?

whitequark, 2024-08-16:
It looks like these are the direct uses of `ClockGen` itself:

```
software/glasgow/applet/audio/dac/__init__.py:66:        m.submodules.clkgen = clkgen = ClockGen(self.pulse_cyc)
software/glasgow/applet/audio/yamaha_opx/__init__.py:181:        m.submodules.clkgen = clkgen = EnableInserter(self.clkgen_ce)(ClockGen(self.master_cyc))
software/glasgow/applet/interface/spi_controller/__init__.py:106:        m.submodules.clkgen = clkgen = ResetInserter(clkgen_reset)(ClockGen(self.period_cyc))
software/glasgow/applet/sensor/hx711/__init__.py:50:            m.submodules.clkgen = clkgen = ClockGen(self.osc_cyc)
```

and these are the indirect uses of `ClockGen.derive`:

```
software/glasgow/applet/audio/dac/__init__.py:179:            pulse_cyc = self.derive_clock(clock_name="modulation",
software/glasgow/applet/audio/dac/__init__.py:181:        sample_cyc = self.derive_clock(clock_name="sampling",
software/glasgow/applet/audio/yamaha_opx/__init__.py:1129:            master_cyc=self.derive_clock(
software/glasgow/applet/interface/spi_controller/__init__.py:345:            period_cyc=self.derive_clock(input_hz=target.sys_clk_freq,
software/glasgow/applet/interface/spi_controller/__init__.py:351:            delay_cyc=self.derive_clock(input_hz=target.sys_clk_freq,
software/glasgow/applet/interface/uart/__init__.py:175:        max_bit_cyc = self.derive_clock(
software/glasgow/applet/interface/uart/__init__.py:216:        manual_cyc = self.derive_clock(
software/glasgow/applet/program/m16c/__init__.py:335:            baud: self.derive_clock(input_hz=target.sys_clk_freq, output_hz=baud)
software/glasgow/applet/sensor/hx711/__init__.py:158:            osc_cyc = self.derive_clock(clock_name="osc",
```

whitequark, 2024-08-16:
All in all it's just five applets total, of which only one (UART) uses `ClockGen.derive` directly. So I guess it's probably fine to go either way.

mbikovitsky, 2024-08-16:
I'll create a PR then.