GlasgowEmbedded/glasgow #688: "E: g.cli: register 0x00 does not exist" when running applet

tinylabs, 2024-09-11:
Just received my glasgow rev C3 yesterday. While walking through the documentation there seems to be an issue running applets. The install went fine. udev rules installed.

```
elliot@workstation:~/projects/glasgow$ glasgow --version
Glasgow 0.1.dev2193+gf2c0ecb (CPython 3.10.12 on Linux-6.9.3-76060903-generic-x86_64-with-glibc2.35 Pop!_OS 22.04 LTS)
elliot@workstation:~/projects/glasgow$ glasgow list
C3-20240821T171012Z
elliot@workstation:~/projects/glasgow$ glasgow run uart -V 3.3 --pin-tx 0 --pin-rx 1 -b 57600 tty
I: g.device.hardware: generating bitstream ID d21f2ed6831a43901015d4b7d4b8eb4e
I: g.cli: running handler for applet 'uart'
E: g.cli: register 0x00 does not exist
```

I can build and flash the uart applet fine.
```
elliot@workstation:~/projects/glasgow_workspace$ glasgow build --rev C3 uart
I: g.cli: generating bitstream for applet 'uart'
elliot@workstation:~/projects/glasgow_workspace$ glasgow flash uart
I: g.cli: reading device configuration
I: g.cli: device has serial C3-20240821T171012Z
I: g.cli: device has flashed firmware
I: g.cli: device has flashed bitstream ID 654b5aaa14e3f1a88970463fd0864462
I: g.cli: generating bitstream for applet uart
I: g.cli: generated bitstream ID 654b5aaa14e3f1a88970463fd0864462
I: g.cli: using built-in firmware
I: g.cli: programming bitstream
I: g.cli: bitstream identical
I: g.cli: programming configuration and firmware
I: g.cli: configuration and firmware identical
```

Any advice would be appreciated. I can't seem to find anyone with a similar issue.

whitequark, 2024-09-11:
This is likely a hardware issue.

1. Try taking the device out of the case if you have it within. Let me know if that helps.
2. Contact @esden for further instructions.

tinylabs, 2024-09-11:
Ok, thank you @whitequark.
Adding some troubleshooting I tried:
- I bypassed the USB3.0 hub that it was plugged into. Didn't help.
- I switch to a different USB-C cable. Didn't help.
- The ACT and ERR LEDs blink fast twice when running the 'uart run' command above.

I'll email @esden to troubleshoot further.

matthewelse, 2024-09-16:
@tinylabs did you manage to find a solution to this? I'm seeing the same errors with ~the same repro

whitequark, 2024-09-16:
We should probably add a message indicating this to be a hardware error--this is ~always a fault on the I2C bus, from my experience.

Could you re-run with `glasgow -vvv ...` to confirm please?

matthewelse, 2024-09-16:
Thanks! This is the output with `-vvv`

```
➜  glasgow git:(main) glasgow -vvv run uart -V 3.3 --pin-tx 0 --pin-rx 1 -b 57600 tty 
D: g.device.hardware: found revC3 device with serial C3-20240821T190159Z
D: g.applet.interface.uart: clock in=48000.000 req=9.600 out=9.600 [kHz] error=0 [ppm] duty=50.0%
D: g.applet.interface.uart: claimed pipe P and port(s) A, B
D: g.applet.interface.uart: adding reset register at address 0x05
D: g.applet.interface.uart: assigning applet port 'rx' to device pin A1
D: g.applet.interface.uart: assigning applet port 'tx' to device pin A0
D: g.target.toolchain: using toolchain 'builtin' (yosys 0.45.0.0.post775, nextpnr-ice40 0.7.0.0.post519, icepack 0.7.0.0.post519)
T: g.target.toolchain: tool 'yosys' is invoked as '/home/melse/.local/share/pipx/venvs/glasgow/bin/yowasp-yosys'
T: g.target.toolchain: tool 'nextpnr-ice40' is invoked as '/home/melse/.local/share/pipx/venvs/glasgow/bin/yowasp-nextpnr-ice40'
T: g.target.toolchain: tool 'icepack' is invoked as '/home/melse/.local/share/pipx/venvs/glasgow/bin/yowasp-icepack'
T: g.target.toolchain: toolchain ID is 0fb575d5e6ecfd3329cec3237d6f0fd7
T: g.target.toolchain: tool ID of 'yosys' is e21c23412a85d1114fb3a09fcee195c2
T: g.target.toolchain: tool ID of 'nextpnr-ice40' is d8e70cbe36257a34e6563bc47eb4c832
T: g.target.toolchain: tool ID of 'icepack' is d8e70cbe36257a34e6563bc47eb4c832
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x18 value=0x0000 index=0x0000 length=16 (submit)
T: g.device.hardware: USB: CONTROL IN data=<87f7eb3795bb06b69417b10dbbdee655> (completed)
I: g.device.hardware: device already has bitstream ID 87f7eb3795bb06b69417b10dbbdee655
I: g.cli: running handler for applet 'uart'
D: g.applet.interface.uart: clock in=48000.000 req=57.600 out=57.623 [kHz] error=400 [ppm] duty=50.0%
T: g.device.hardware: register 0 write: 0x340
T: g.device.hardware: USB: CONTROL OUT type=0x40 request=0x13 value=0x0000 index=0x0000 data=<00000340> (submit)
T: g.device.hardware: USB: CONTROL IN type=0x40 request=0x12 value=0x0000 index=0x0000 length=1 (submit)
T: g.device.hardware: USB: CONTROL IN data=<02> (completed)
E: g.cli: register 0x00 does not exist
```

tinylabs, 2024-09-16:
> @tinylabs did you manage to find a solution to this? I'm seeing the same errors with ~the same repro

It appears to be a hardware issue. After working with @esden I ended up RMA'ing my unit.

whitequark, 2024-09-21:
Changes in #699 should address the issue.

closed 2024-09-21 by whitequark