GlasgowEmbedded/glasgow #726: Building firmware fails

twam, 2024-12-02:
I tried to build the firmware as described on http://glasgow-embedded.org/latest/develop/firmware.html with macOS and Gentoo Linux. On both systems it fails with:

```
$ make -C vendor/libfx2/firmware
make: Entering directory '/home/twam/glasgow/vendor/libfx2/firmware'
make[1]: Entering directory '/home/twam/glasgow/vendor/libfx2/firmware/library'
make[1]: Leaving directory '/home/twam/glasgow/vendor/libfx2/firmware/library'
make[1]: Entering directory '/home/twam/glasgow/vendor/libfx2/firmware/boot-cypress'
make[1]: *** No rule to make target 'build/main.rel', needed by 'boot-cypress.ihex'.  Stop.
make[1]: Leaving directory '/home/twam/glasgow/vendor/libfx2/firmware/boot-cypress'
make: *** [Makefile:4: all] Error 2
make: Leaving directory '/home/twam/glasgow/vendor/libfx2/firmware'
```

Any hints how to solve it?

whitequark, 2024-12-02:
That's odd. Does `./software/deploy-firmware.sh` work?

twam, 2024-12-02:
I don't have a running docker setup for now. :)

I found out, that `sdcc` list the following changes in the manual:

>• In 4.2.9, support for arguments to -MMD was dropped. The output file can be specified via -MF instead.

and I'm running `sdcc` 4.4.0. This explains why

```
 sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x0200 --std-sdcc99 --model-small  -I../library/include -L../library/lib/small -Wp,-MQ,build/main.rel,-MMD,build/main.d -c main.c
```
always deleted `main.c` resulting in the described the failure. Replacing the `-MMD` with `-M,-MF` makes the firmware build, but with a lot of
```
at 1: warning 190: ISO C forbids an empty translation unit
```
Flashing the created (almost empty 🙈) `firmware.ihex` bricked my Glasgow :)

xeniter, 2025-02-09:
had exact same error, thanks for fix works now with latest libfx2

to get it you have to run:
git submodule update --remote

made request for update to latest version:
https://github.com/GlasgowEmbedded/glasgow/pull/739

whitequark, 2025-02-09:
Fixed by #739.

closed 2025-02-09 by whitequark