GlasgowEmbedded/glasgow #748: applet.internal.benchmark libusb1 `ValueError` under PyPy

Sazzach, 2025-03-02:
Running the benchmark applet with PyPy results in either a ValueError exception, or the applet hanging indefinitely. I noticed this issue you raised here: vpelletier/python-libusb1/issues/98, and changed the libusb1 version mentioned in the thread (vpelletier/python-libusb1@11e700bcbebceff3752a34849a39a360d0af4322) which fixed it.

I'm not sure if you want to wait for a libusb1 release on pypi including the finalizer lock or not so I made this issue.

Glasgow version: `Glasgow 0.1.dev2243+ga667a26 (PyPy 3.10.14 on Linux-6.8.0-54-generic-x86_64-with-glibc2.39 Ubuntu 24.04.2 LTS)`

The full debug output is ~140MB, the following is the tail end of it.
```
T: glasgow.device.hardware: USB: BULK EP6 IN length=16384 (submit)
T: glasgow.applet.internal.benchmark: FIFO: flush
T: glasgow.applet.internal.benchmark: FIFO: wait for flush
T: glasgow.applet.internal.benchmark: FIFO: cancelling operations
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
T: glasgow.device.hardware: USB: BULK EP6 IN (cancelled)
Traceback (most recent call last):
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/.venv/bin/glasgow", line 8, in <module>
    sys.exit(run_main())
             ^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/cli.py", line 944, in run_main
    exit(asyncio.new_event_loop().run_until_complete(main()))
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/futures.py", line 201, in result
    raise self._exception.with_traceback(self._exception_tb)
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/tasks.py", line 232, in __step
    result = coro.send(None)
             ^^^^^^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/cli.py", line 712, in main
    return applet_task.result()
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/futures.py", line 201, in result
    raise self._exception.with_traceback(self._exception_tb)
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/tasks.py", line 232, in __step
    result = coro.send(None)
             ^^^^^^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/cli.py", line 659, in run_applet
    return await applet.interact(device, args, iface)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/applet/internal/benchmark/__init__.py", line 212, in interact
    actual = await iface.read(len(packetmax))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/access/direct/demultiplexer.py", line 258, in read
    await self._in_tasks.wait_one()
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/support/task_queue.py", line 84, in wait_one
    await self.poll()
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/support/task_queue.py", line 71, in poll
    await self._done.popleft()
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/futures.py", line 288, in __await__
    return self.result()  # May raise too.
           ^^^^^^^^^^^^^
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/futures.py", line 201, in result
    raise self._exception.with_traceback(self._exception_tb)
  File "/home/samuelz/.local/share/pdm/python/pypy@3.10.14/lib/pypy3.10/asyncio/tasks.py", line 232, in __step
    result = coro.send(None)
             ^^^^^^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/access/direct/demultiplexer.py", line 233, in _in_task
    data = await self.device.bulk_read(self._endpoint_in, size)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/device/hardware.py", line 344, in bulk_read
    data = await self._do_transfer(is_read=True, setup=lambda transfer:
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/glasgow/device/hardware.py", line 265, in _do_transfer
    transfer = self.usb_handle.getTransfer()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/.venv/lib/pypy3.10/site-packages/usb1/__init__.py", line 1517, in getTransfer
    return USBTransfer(
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/.venv/lib/pypy3.10/site-packages/usb1/__init__.py", line 257, in __init__
    registerFinalizer(finalizer_handle, self.__close)
  File "/home/samuelz/Documents/glasgow_devel/glasgow3/glasgow/software/.venv/lib/pypy3.10/site-packages/usb1/__init__.py", line 1047, in __registerFinalizer
    raise ValueError('Finalizer handle {handle} already exists')
ValueError: Finalizer handle {handle} already exists
```

whitequark, 2025-03-02:
Thanks, I completely forgot.

Sazzach, 2025-03-12:
I tried it with the new libusb1 release(`libusb1 3.3.0`) and it is fixed with that version too.

One interesting thing to note is that the USB performance as measured by the benchmark with `pypy` is substantially lower than `cpython`.

pypy:
```
> pdm run glasgow run benchmark
I: g.device.hardware: device already has bitstream ID 4c6d76f0db309e9240e75479026105cb
I: g.cli: running handler for applet 'benchmark'
I: g.applet.internal.benchmark: running benchmark mode source for 8.000 MiB
I: g.applet.internal.benchmark: mode source: 10.74 MiB/s (85.92 Mb/s)
I: g.applet.internal.benchmark: running benchmark mode sink for 8.000 MiB
I: g.applet.internal.benchmark: mode sink: 14.62 MiB/s (116.99 Mb/s)
I: g.applet.internal.benchmark: running benchmark mode loopback for 8.000 MiB
I: g.applet.internal.benchmark: mode loopback: 17.18 MiB/s (137.41 Mb/s)
I: g.applet.internal.benchmark: running benchmark mode latency for 8.000 MiB
I: g.applet.internal.benchmark: mode latency: mean: 956.01 µs stddev: 2003.66 µs worst: 89243.84 µs
```

cpython:
```
> pdm run glasgow run benchmark
I: g.device.hardware: device already has bitstream ID 4c6d76f0db309e9240e75479026105cb
I: g.cli: running handler for applet 'benchmark'
I: g.applet.internal.benchmark: running benchmark mode source for 8.000 MiB
I: g.applet.internal.benchmark: mode source: 41.09 MiB/s (328.75 Mb/s)
I: g.applet.internal.benchmark: running benchmark mode sink for 8.000 MiB
I: g.applet.internal.benchmark: mode sink: 35.55 MiB/s (284.37 Mb/s)
I: g.applet.internal.benchmark: running benchmark mode loopback for 8.000 MiB
I: g.applet.internal.benchmark: mode loopback: 41.74 MiB/s (333.88 Mb/s)
I: g.applet.internal.benchmark: running benchmark mode latency for 8.000 MiB
I: g.applet.internal.benchmark: mode latency: mean: 530.03 µs stddev: 71.61 µs worst: 1769.94 µs
```

whitequark, 2025-03-12:
> One interesting thing to note is that the USB performance as measured by the benchmark with pypy is substantially lower than cpython.

I'm afraid I don't have the capacity to investigate that for foreseeable future.

Sazzach, 2025-03-12:
>I'm afraid I don't have the capacity to investigate that for foreseeable future.

No worries! I didn't intend that to come across as a request for you to investigate. As I was testing, I noticed the performance difference and figured that it was notable / could potentially signify another issue wrt PyPy and libusb1. I'm not sure that is actually the case, but I didn't want to sit on a potential issue.

commit 12803780 (2025-04-26): software: bump libusb minimum version.

Fixes #748 (crash on PyPy).
PR #785 (2025-04-26): Bump libusb minimum version
Fixes #748 (crash on PyPy).
closed 2025-04-26 by whitequark