GlasgowEmbedded/glasgow #751: Evaluate Corebai CBM9002A as a replacement for Cypress CY7C68013A

whitequark, 2025-03-11:
The FX2 is [being discontinued on extremely short notice](https://www.mouser.com/PCN/Infineon_cPD_314_24.pdf), with the last order date being this Saturday, and we have no other replacement for it. If the Corebai ASIC does not work we have to retire revA..revC designs entirely.

ewenmcneill, 2025-03-11:
Looking closer at the Product Discontinuation notice I see it's actually dated "2024-09-12" (bottom of each page), so coming up 6 months ago.   I guess we're learning about it now because resellers (eg, DigiKey / Mouser) are trying to finalise *their* last orders.

Also of note, the disclaimer at the bottom says: "Final orders can only be accepted on or before the last order date and are **subject to the availability**.  Early ordering is strongly recommended" (emphasis added)

So it seems like this might *not* be related to Skywater buying the Infineon Austin fab (February 2025)?   Possibly Cypress had it on an older Skywater process node, which got discontinued?  (I vaguely remember hearing Skywater were decomissioning an old process node a while back, but I'm not quickly finding it.)

Anyway, the actions are still "maybe buy some more stock" and "find a replacement for new boards".

Ewen

whitequark, 2025-03-11:
> So it seems like this might _not_ be related to Skywater buying the Infineon Austin fab (February 2025)?

Deals like this are usually done months in advance so I wouldn't be surprised if it is.

However, Andrew Zonenberg provided a plausible seeming explanation for it: newer product lines (CYUSB*) are simply higher margin so Infineon wants us to buy that, NRE be damned.

whitequark, 2025-03-25:
Thanks to the emergency testing efforts of @AleksaBjelogrlic we now know that, at the very least, basic functions of the device work fine (`benchmark source`, `benchmark sink`, `uart` applets). So, while there is always risk involved with a part replacement like that, this is very encouraging as it suggests there are not any major issues and the timing on the FX2 bus is fine.

Therefore, I will recommend that anyone having trouble sourcing the FX2 use the Corebai chip instead.

CounterPillow, 2025-03-26:
A replacement was seemingly already found, but I'm gonna leave this possible alternative here anyway
https://www.wch-ic.com/products/CH569.html (I have an evaluation kit of this, but haven't found the time to play with it yet)

100+ unit pricing on LCSC is $5.0353 per chip.

sd-fritze, 2025-03-26:
While that's a part that looks interesting capabilities-wise, the datasheet contains
"Please refer to and call the provided subroutine library for specific applications."
far too often (for most high-speed peripherals) - and some of the others are extremely underdocumented :(

--gruetzkopf

whitequark, 2025-07-24:
Based on an evaluation by a contract manufacturer I talked to, CBM9002A will remain in production until at least 2030. As such I believe it's an adequate replacement for the FX2LP and we can use it for foreseeable future.

closed 2025-07-24 by whitequark

GerdchenF, 2025-08-22:
> Thanks to the emergency testing efforts of [@AleksaBjelogrlic](https://github.com/AleksaBjelogrlic) we now know that, at the very least, basic functions of the device work fine (`benchmark source`, `benchmark sink`, `uart` applets).

Sorry for writing on a closed matter, but are details of the test available? I didn't find anything on that either here or at Aleksa's page.