GlasgowEmbedded/glasgow #794: Get rid of explicit flush / autoflush as currently implemented

whitequark, 2025-05-01:
Right now, flow control works as follows. If you have a complete 512-byte packet, it is immediately submitted to the USB host. If the buffer has less than 512 bytes, you have one of two options:
1. Autoflush enabled (default): whenever the applet is no longer actively writing to the buffer, even for one cycle, the buffer is submitted.
2. Explicit flush (`auto_flush=False`): whenever the applet is asserting a signal, the buffer is submitted; otherwise it is never submitted.

Having (2) as the default caused a lot of problems early on where applets would deadlock because a flush operation was forgotten somewhere. Having (1) is a default is problematic now because it has really high performance (bandwidth, latency, *and* CPU time) overhead.

Instead of (1) by default we should use a version of [Nagle's algorithm](https://en.wikipedia.org/wiki/Nagle's_algorithm): submit a packet if it's been sitting in the buffer for a few milliseconds. We probably will also need an equivalent of `TCP_NODELAY`; although the delay is imperceptible on human timescales, protocols that require thousands of roundtrips per second will still visibly suffer from this change. However, there are many protocols which primarily transfer latency-insensitive bulk data, and the implementations of them would benefit from a default that is easier to work with.

urjaman, 2025-05-10:
I think an optionally adjustable inter-byte (or inter-write, regardless of unit size, i assume we're talking bytes) timeout (couple of milliseconds, Nagle-like) _and_ a total packet timeout (like 20ms / 50Hz) would serve most users quite well.

This thought provided to you by me going "instead of NODELAY, how about we just have adjustable delay that can be set to 0, that way things with a baud rate (or similar) could set their expected speed" - but then a 19200 baud UART will buffer for 0.27 seconds and dump the full 512 bytes, that's not optimal either - so in those cases sending packets in the handful of bytes size (1920 / 50 = 38.4) would be optimal.

whitequark, 2025-05-10:
What is the difference between "inter-write timeout" and "total packet timeout"?

urjaman, 2025-05-10:
total packet timeout counts from (resets at) the first write to the packet buffer but not on subsequent writes until the packet is sent (by being full, inter-byte timeout or the total packet timeout).

(Actually now that I'm thinking, alternatively you could just compute a more suitable maximum packet size, that might be cheaper on the gateware.)

whitequark, 2025-07-12:
@urjaman I've solved this in a slightly different manner in https://github.com/GlasgowEmbedded/glasgow/pull/948. The problem is that asserting PKTEND on a maximum-length packet may flush it to the HC, but it won't complete the URB. So the "use the bus efficiently" timeout and the "deliver data promptly" timeout can't use the same mechanism.

What do you think? It seems to work well in my testing.