GlasgowEmbedded/glasgow #975: software/deploy-firmware does not reproduce on aarch64-linux

flokli, 2025-07-27:
I ran `./software/deploy-firmware.sh` (with https://github.com/GlasgowEmbedded/glasgow/pull/974 applied too). It successfully built, but the produced `software/glasgow/device/firmware.ihex` file was different afterwards.

Build log:

```
❯ ./software/deploy-firmware.sh
+ apt-get update -qq
+ apt-get install -qq --no-install-recommends git make sdcc
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package perl-modules-5.36.
(Reading database ... 6083 files and directories currently installed.)
Preparing to unpack .../00-perl-modules-5.36_5.36.0-7+deb12u2_all.deb ...
Unpacking perl-modules-5.36 (5.36.0-7+deb12u2) ...
Selecting previously unselected package libgdbm6:arm64.
Preparing to unpack .../01-libgdbm6_1.23-3_arm64.deb ...
Unpacking libgdbm6:arm64 (1.23-3) ...
Selecting previously unselected package libgdbm-compat4:arm64.
Preparing to unpack .../02-libgdbm-compat4_1.23-3_arm64.deb ...
Unpacking libgdbm-compat4:arm64 (1.23-3) ...
Selecting previously unselected package libperl5.36:arm64.
Preparing to unpack .../03-libperl5.36_5.36.0-7+deb12u2_arm64.deb ...
Unpacking libperl5.36:arm64 (5.36.0-7+deb12u2) ...
Selecting previously unselected package perl.
Preparing to unpack .../04-perl_5.36.0-7+deb12u2_arm64.deb ...
Unpacking perl (5.36.0-7+deb12u2) ...
Selecting previously unselected package libbrotli1:arm64.
Preparing to unpack .../05-libbrotli1_1.0.9-2+b6_arm64.deb ...
Unpacking libbrotli1:arm64 (1.0.9-2+b6) ...
Selecting previously unselected package libkrb5support0:arm64.
Preparing to unpack .../06-libkrb5support0_1.20.1-2+deb12u3_arm64.deb ...
Unpacking libkrb5support0:arm64 (1.20.1-2+deb12u3) ...
Selecting previously unselected package libk5crypto3:arm64.
Preparing to unpack .../07-libk5crypto3_1.20.1-2+deb12u3_arm64.deb ...
Unpacking libk5crypto3:arm64 (1.20.1-2+deb12u3) ...
Selecting previously unselected package libkeyutils1:arm64.
Preparing to unpack .../08-libkeyutils1_1.6.3-2_arm64.deb ...
Unpacking libkeyutils1:arm64 (1.6.3-2) ...
Selecting previously unselected package libssl3:arm64.
Preparing to unpack .../09-libssl3_3.0.16-1~deb12u1_arm64.deb ...
Unpacking libssl3:arm64 (3.0.16-1~deb12u1) ...
Selecting previously unselected package libkrb5-3:arm64.
Preparing to unpack .../10-libkrb5-3_1.20.1-2+deb12u3_arm64.deb ...
Unpacking libkrb5-3:arm64 (1.20.1-2+deb12u3) ...
Selecting previously unselected package libgssapi-krb5-2:arm64.
Preparing to unpack .../11-libgssapi-krb5-2_1.20.1-2+deb12u3_arm64.deb ...
Unpacking libgssapi-krb5-2:arm64 (1.20.1-2+deb12u3) ...
Selecting previously unselected package libsasl2-modules-db:arm64.
Preparing to unpack .../12-libsasl2-modules-db_2.1.28+dfsg-10_arm64.deb ...
Unpacking libsasl2-modules-db:arm64 (2.1.28+dfsg-10) ...
Selecting previously unselected package libsasl2-2:arm64.
Preparing to unpack .../13-libsasl2-2_2.1.28+dfsg-10_arm64.deb ...
Unpacking libsasl2-2:arm64 (2.1.28+dfsg-10) ...
Selecting previously unselected package libldap-2.5-0:arm64.
Preparing to unpack .../14-libldap-2.5-0_2.5.13+dfsg-5_arm64.deb ...
Unpacking libldap-2.5-0:arm64 (2.5.13+dfsg-5) ...
Selecting previously unselected package libnghttp2-14:arm64.
Preparing to unpack .../15-libnghttp2-14_1.52.0-1+deb12u2_arm64.deb ...
Unpacking libnghttp2-14:arm64 (1.52.0-1+deb12u2) ...
Selecting previously unselected package libpsl5:arm64.
Preparing to unpack .../16-libpsl5_0.21.2-1_arm64.deb ...
Unpacking libpsl5:arm64 (0.21.2-1) ...
Selecting previously unselected package librtmp1:arm64.
Preparing to unpack .../17-librtmp1_2.4+20151223.gitfa8646d.1-2+b2_arm64.deb ...
Unpacking librtmp1:arm64 (2.4+20151223.gitfa8646d.1-2+b2) ...
Selecting previously unselected package libssh2-1:arm64.
Preparing to unpack .../18-libssh2-1_1.10.0-3+b1_arm64.deb ...
Unpacking libssh2-1:arm64 (1.10.0-3+b1) ...
Selecting previously unselected package libcurl3-gnutls:arm64.
Preparing to unpack .../19-libcurl3-gnutls_7.88.1-10+deb12u12_arm64.deb ...
Unpacking libcurl3-gnutls:arm64 (7.88.1-10+deb12u12) ...
[... 137 lines cut ...]
sdcc -mmcs51 --std-sdcc99 -Iinclude  --model-medium -c -o build/medium/uf2fat.rel uf2fat.c
sdar rc lib/medium/fx2uf2.lib build/medium/uf2scsi.rel build/medium/uf2fat.rel
make: Leaving directory '/glasgow/vendor/libfx2/firmware/library'
make: Entering directory '/glasgow/firmware'
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/main.rel,-MMD,build/main.d -c -o build/main.rel main.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/fpga.rel,-MMD,build/fpga.d -c -o build/fpga.rel fpga.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/dac_ldo.rel,-MMD,build/dac_ldo.d -c -o build/dac_ldo.rel dac_ldo.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/adc_adc081c.rel,-MMD,build/adc_adc081c.d -c -o build/adc_adc081c.rel adc_adc081c.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/adc_ina233.rel,-MMD,build/adc_ina233.d -c -o build/adc_ina233.rel adc_ina233.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/pull.rel,-MMD,build/pull.d -c -o build/pull.rel pull.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/fifo.rel,-MMD,build/fifo.d -c -o build/fifo.rel fifo.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -Wp,-MQ,build/util.rel,-MMD,build/util.d -c -o build/util.rel util.c
sdcc -mmcs51 --iram-size 0x100 --code-size 0x3e00 --xram-loc  0x3e00 --xram-size 0x01c0 --std-sdcc99 --model-medium -DSYNCDELAYLEN=16 -DCONF_SIZE=0x0040 -I../vendor/libfx2/firmware/library/include -L../vendor/libfx2/firmware/library/lib/medium -o build/glasgow.ihex build/main.rel build/fpga.rel build/dac_ldo.rel build/adc_adc081c.rel build/adc_ina233.rel build/pull.rel build/fifo.rel build/util.rel ../vendor/libfx2/firmware/library/lib/medium/fx2.lib ../vendor/libfx2/firmware/library/lib/medium/fx2isrs.lib ../vendor/libfx2/firmware/library/lib/medium/fx2usb.lib
make: Leaving directory '/glasgow/firmware'
❯ gst
HEAD detached at 2a67f79d
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   software/glasgow/device/firmware.ihex

no changes added to commit (use "git add" and/or "git commit -a")
```

This might be due to me running it on aarch64-linux, not x86_64-linux.

flokli, 2025-07-27:
On a clean checkout, and on an x86_64-linux machine, I also came up with the same firmware from above (md5sum 032edb8d077268c780b72a006c8d8b59), but not with the one checked into the repo (9585d689aa9d0f8805c5e33db8ade2cb).

whitequark, 2025-07-28:
> On a clean checkout, and on an x86_64-linux machine, I also came up with the same firmware from above (md5sum 032edb8d077268c780b72a006c8d8b59), but not with the one checked into the repo (9585d689aa9d0f8805c5e33db8ade2cb).

I'm confused. At commit 27a68fc6a2fec37a22621d340c344990d06a4a3b (current HEAD), I get an md5sum `7b61f4ceba2355cda045b3a923b3889d` and `67716b68c9ab779d21d462e235567b4991ba7f35`. Where are you getting the file with the md5sum `9585d689aa9d0f8805c5e33db8ade2cb`?

whitequark, 2025-07-28:
OK, I can tell from the firmware file that it corresponds to the git commit b78ec447 instead:

```
commit b78ec447e285779838f16eadc901ddd990420fe4
Author: Purdea Andrei <andrei@purdea.ro>
Date:   Sat Aug 3 03:10:42 2024 +0300
```

That's a commit from a year ago, not current HEAD. I think the fact that I can do this is a fantastic argument against your proposal in #977.

flokli, 2025-07-28:
I had 427b29da1243dc1a6e105767b13f54e88e401021 checked out, with https://github.com/GlasgowEmbedded/glasgow/pull/974 cherry-picked on top, which happened to have not a changed `firmware` directory since b78ec447e285779838f16eadc901ddd990420fe4 (so that's the commit `git log firmware` returned for me).

closed 2025-07-28 by flokli