GlasgowEmbedded/glasgow #982: Unable to upgrade firmware from 2025-01-26

flokli, 2025-07-28:
I updated glasgow from 2a67f79d6025a06e98277956cbb036c4237960f1 to 8a61053c7b7ff241c096eba68c6ab8fc0e0389f7. 

I got prompted to upgrade the firmware, but had issues with `glasgow flash` running into `LIBUSB_ERROR_TIMEOUT`.

I get one of the two error behaviours during a `glasgow flash` invocation:

```
sudo result/bin/glasgow --log-file issue.log flash
I: g.hardware.device: found revC3 device with API level 3 (supported API level is 4)
Traceback (most recent call last):
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/support/usb/libusb1.py", line 58, in wrapper
    return f(*args, **kwargs)
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/support/usb/libusb1.py", line 205, in serial_number
    self._serial_number = self._ensure_open.getASCIIStringDescriptor(
                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        self._impl_device.getSerialNumberDescriptor())
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/bpqzv0gm5wcgxx7j637gh4wf9wjr86wj-python3.13-libusb1-3.3.1/lib/python3.13/site-packages/usb1/__init__.py", line 1364, in getASCIIStringDescriptor
    received = mayRaiseUSBError(libusb1.libusb_get_string_descriptor_ascii(
        self.__handle, descriptor,
        create_binary_buffer(descriptor_string)[0],
        STRING_LENGTH,
    ))
  File "/nix/store/bpqzv0gm5wcgxx7j637gh4wf9wjr86wj-python3.13-libusb1-3.3.1/lib/python3.13/site-packages/usb1/__init__.py", line 129, in mayRaiseUSBError
    __raiseUSBError(value)
    ~~~~~~~~~~~~~~~^^^^^^^
  File "/nix/store/bpqzv0gm5wcgxx7j637gh4wf9wjr86wj-python3.13-libusb1-3.3.1/lib/python3.13/site-packages/usb1/__init__.py", line 121, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorTimeout: LIBUSB_ERROR_TIMEOUT [-7]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/bin/.glasgow-wrapped", line 9, in <module>
    sys.exit(run_main())
             ~~~~~~~~^^
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/cli.py", line 1056, in run_main
    exit(asyncio.new_event_loop().run_until_complete(main()))
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/nix/store/chgczp3c2zsr14bxkx9vfwsywajbgbgg-python3-3.13.5/lib/python3.13/asyncio/base_events.py", line 725, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/cli.py", line 605, in main
    device = await GlasgowDevice.find(args.serial)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/hardware/device.py", line 187, in find
    usb_devices = await cls._enumerate_devices(usb_context)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/hardware/device.py", line 114, in _enumerate_devices
    device.serial_number)
    ^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/support/usb/libusb1.py", line 60, in wrapper
    map_error(err)
    ~~~~~~~~~^^^^^
  File "/nix/store/3w5rxhwl847595k46dxvqsklqrcdgcas-glasgow-0-unstable-2025-07-24/lib/python3.13/site-packages/glasgow/support/usb/libusb1.py", line 45, in map_error
    raise Error(str(error)) from error
glasgow.support.usb.Error: LIBUSB_ERROR_TIMEOUT [-7]
```

```
I: g.hardware.device: found revC3 device with API level 3 (supported API level is 4)
W: g.hardware.device: please run `glasgow flash` to update firmware of device C3-20241011T135145Z
W: g.hardware.device: device 001/002 did not re-enumerate after firmware upload
E: g.cli: device not found
```

I was only able to reproduce when having the older firmware present; running `glasgow flash --remove-firmware` with the new glasgow version was not sufficient.

I reproduced this by building nixpkgs' on current `master` (which uses the upstream `.ihex` file due to a packaging bug), and by building https://github.com/NixOS/nixpkgs/pull/382355 (which produces the same firmware as upstream), by invoking `sudo $(nix-build -A glasgow)/bin/glasgow flash` with a commit before and after the bump to `0-unstable-2025-07-28`. [^1]

[^1]: Contrary to the older reports in https://github.com/GlasgowEmbedded/glasgow/pull/958, I was also able to reproduce the issues with the parent commit of that PR.

flokli, 2025-07-28:
Running
 - `glasgow flash --remove-firmware` with 2a67f79d6025a06e98277956cbb036c4237960f1,
 - replug
 - `glasgow flash` with 42327220a33f70b061c9103309364b6ecc1c507f
 -  replug
 - `glasgow flash` with 8a61053c7b7ff241c096eba68c6ab8fc0e0389f7
 -  replug

seems to be a way to workaround this issue and lift the device to API level 4.

If I skip step the  `glasgow flash` with 42327220a33f70b061c9103309364b6ecc1c507f, replug, then
`glasgow flash` with 8a61053c7b7ff241c096eba68c6ab8fc0e0389f7 shows the following error:

```
W: g.hardware.device: device 001/003 did not re-enumerate after firmware upload
E: g.cli: device not found
```
So it looks like somewhere between 42327220a33f70b061c9103309364b6ecc1c507f..8a61053c7b7ff241c096eba68c6ab8fc0e0389f7 `glasgow flash` lost the ability to deal with unflashed, or API level 3 devices.

flokli, 2025-07-28:
42327220a33f70b061c9103309364b6ecc1c507f (July 18, 2025) already contains the fix b78ec447e285779838f16eadc901ddd990420fe4 (Aug 3, 2024).

whitequark, 2025-07-28:
OK, I have simplified your reproducer. On current HEAD:

```
$ glasgow flash --remove-firmware
# replug
$ glasgow list
```

is sufficient to cause a timeout. It is not sporadic but rather reliably occurs every time.

whitequark, 2025-07-28:
Fixed in https://github.com/GlasgowEmbedded/glasgow/pull/983. I made an error while refactoring and wrong code was being uploaded.