GlasgowEmbedded/glasgow #1019: Schematics.pdf for Rev C3 shows "Rev C2" in the page header.

langleyVa, 2025-09-24:
### Version

n.A.

### Description

The schematic.pdf for rev C3 (see https://github.com/GlasgowEmbedded/glasgow/blob/main/hardware/boards/glasgow/revC3/schematics.pdf) says "Rev: C2" in the lower right corner of every sheet.

whitequark, 2025-09-24:
Thanks. Could you submit a PR fixing this?

langleyVa, 2025-09-24:
Well, sorry about this, but:

The version of Kicad that Ubuntu gives me (7.0.11+dfsg-1build4) is newer than what is listed in the PDF (7.0.8-7.0.8~ubuntu22.04.1). It still refuses to open the schematic: "KiCad was unable to open this file because it was created with a more recent version than the one you are running. To open it you will need to upgrade KiCad to a version dated 20231120 or later."

The version that the Kicad PPA gives me (9.0.4) changes a few bits and bobs in the project files when saving, but also changes all instances of "+3.3V" to "+3V3". That's way out of my comfort zone, it seems like a great way to screw something up in subtle ways that I don't know enough about to make sure that I don't mess them up.

I can offer a pull request where I just commit whatever Kicad 9.0.4 decides is the best way forward for the project, or I have to pass this on to someone who has the right version of Kicad lying around.

whitequark, 2025-09-24:
@q3k Do you think you could handle this?

langleyVa, 2025-09-24:
P.S. Last Commit message on the .kicad_sch file says "Open and save with Kicad 8.0.9"

q3k, 2025-09-24:
See #1020

langleyVa, 2025-09-25:
That closed it, thank you!

closed 2025-09-25 by langleyVa